cilynx/rtl88x2bu #1: How to completely remove and reinstall RTL88x2BU drivers in Ubuntu MATE 19.04

TheJags, 2019-10-10:
This driver worked just fine for 5 days, and now suddenly stopped working. Now my `TP-Link Archer T3U dual-band USB wireless adapter` is not even recognized at all. I tried reinstalling the driver but getting an error that said module already installed (or a similar message).

How could I completely remove/uninstall kernel modules for RTL88x2BU please... so I could then try reinstalling. Many thanks.

````
Kernel version: 5.3.4
OS: Ubuntu MATE 19.04 64-bit
````

ranjan-purbey, 2019-11-06:
If you used dkms to install the driver, follow these steps to uninstall the driver:
1. Find the version of driver installed on your system
```
$ dkms status
rtl88x2bu, *5.6.1*, 5.3.0-19-generic, x86_64: installed
```
  _here *5.6.1* is the installed driver version_

2. Uninstall the driver (driver version might be different in your case)
```
$ sudo dkms uninstall rtl88x2bu/5.6.1
```
3. Remove the driver from dkms tree
```
$ sudo dkms remove rtl88x2bu/5.6.1 --all
```

If you get any errors, please share the logs. Otherwise kindly close the issue.

TheJags, 2019-11-09:
@ranjan-purbey thanks for the reply. I was able to remove and reinstall the drivers with the help from this AskUbuntu thread:
````
https://askubuntu.com/questions/1182236/how-to-properly-remove-kernel-modules-for-rtl88x2bu-in-ubuntu-mate-19-04
````

closed 2019-11-09 by TheJags