cilynx/rtl88x2bu #2: always required autenthication without connect

pipes80-collab, 2019-11-09:
I have installed fine ur repo but when i try to connect at the wifi network ask password and try to connect and reask the password

cilynx, 2019-11-14:
Which wifi adapter, linux flavor, and kernel version?

pipes80-collab, 2019-11-14:
this is the adapter Realtek 8812BU Wireless LAN 802.11ac USB NIC
, I use debian Sid with kernel 5.3.0-2 and patches 5.3.9-2

gsalgado, 2019-11-14:
Do you have an entry for your adapter's MAC address in `/etc/udev/rules.d/70-persistent-net.rules` (or similar)? I believe the lack of that would cause the interface to have a different name on every boot and that would in turn cause network-manager to ask for the password again

pipes80-collab, 2019-11-14:
no i have not do that

pipes80-collab, 2019-11-25:
RTL 8812bu usb