cilynx/rtl88x2bu #5: New 5.6.1.5 source available

JWSmythe, 2020-01-01:
Inside this file: 
http://files.dlink.com.au/Products/DWA-182/REV_D/Drivers/DWA-182_D1_Linux_Driver_v5.6.1.5.zip

Is the 5.6.1.5 source:
20190718_RTL88x2BU_WiFi_linux_v5.6.1.5_33902.20190604_COEX20180928-6a6a-0718.tar.bz2

Unfortunately, 5.6.1.5 doesn't compile as-is on my Linux 5.4.6 kernel system.

I wasn't sure how to send this privately, so you can just delete this after you've read it.

Enjoy.

jasonbraganza, 2020-01-02:
Thank you so much :)

z419, 2020-01-10:
I've been testing this for the past day as an AP with WPA3 enabled. It's been workin

z419, 2020-01-10:
[rtl88x2bu-5.6.1.5_linux-4.19+.patch](https://gist.github.com/z419/d88f690360baa455b190641f92079cfb)

I've been testing this for the past day in AP mode with WPA3. It's working great.

cilynx, 2020-04-08:
New branch created:
https://github.com/cilynx/rtl88x2bu/tree/5.6.1.6_35492.20191025_COEX20180928-6a6a

Just starting cleanup, but feel free to test.  I confirmed it compiles on 5.4, but it has many of the old usual suspect compiler warnings.

Once we get it cleaned up and have a few users confirm functionality, we'll make that the default branch on the repo.

closed 2020-04-08 by cilynx