cilynx/rtl88x2bu #6: Still not working on Ubuntu 19.10 :-(

Polve, 2020-01-13:
The driver compiles and installs fine, but then it doesn't work: it detects the networks and sometimes it's also able to connect, but the connection does not work, and after a few minutes the connection is lost.

gsedubun, 2020-01-23:
what device did you use?

powderluv, 2020-01-30:
works on my 20.04 beta install (using as a hostap)
wlx74ee2ae24062  IEEE 802.11an  ESSID:"borg"  Nickname:"<WIFI@REALTEK>"
          Mode:Master  Frequency:5.18 GHz  Access Point: 74:EE:2A:E2:40:62   
          Bit Rate:144.4 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=0/100  Signal level=-100 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

nxadm, 2020-02-19:
I just installed this device on Ubuntu 19.10 without any problem:
https://www.amazon.fr/gp/product/B07QR5GD7G/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

gsedubun, 2020-02-20:
i use [tp-link t4u archer v3 ](https://www.tp-link.com/us/support/download/archer-t4u/)
works fine with this driver.

MaxG87, 2020-04-04:
I have a D-Link DWA-182 and it does not work on Ubuntu 19.10 using this driver here. I suspect that it does not work on any kernel >=5.3. Issue #27 might be relevant too.

powderluv, 2020-04-04:
I have used it on 19.10 and 20.04 beta. What is the error?  What is the PCI ID of the device?

MaxG87, 2020-04-04:
A detailed bug report is on my related question [on StackOverflow](https://stackoverflow.com/questions/60832869/self-compiled-linux-driver-stopped-working-after-upgrading-debian).
The report of `lsusb -v` is:

```
Bus 002 Device 007: ID 2001:331c D-Link Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x2001 D-Link Corp.
  idProduct          0x331c 
  bcdDevice            2.10
  iManufacturer           1 
  iProduct                2 802.11ac NIC
  iSerial                 3 123456
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0035
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 802.11ac NIC
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      HIRD Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x0006
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat        1023 micro seconds
Device Status:     0x0000
  (Bus Powered)
```

powderluv, 2020-04-04:
That PCI device is supported by the driver. BUT looks like you are using some old module.

From your stackoverflow:

[   98.065544] 88x2bu: loading out-of-tree module taints kernel.
[   98.067018] 88x2bu: module verification failed: signature and/or required key missing - tainting kernel
[   98.093187] RTW: module init start
[   98.093189] RTW: rtl88x2bu v5.3.1_27678.20180430_COEX20180427-5959

You want to be using the code from this repo which should be 5.6.1+ 

I would suggest a full removal of the dkms and re-install. I have it running on my Thinkpad + 20.04 with no problems. 

Hope it helps.

MaxG87, 2020-04-07:
I checked again and it worked now. I am very sorry. I did so much debugging on this issue, I obviously lost track of what I did which machine and with which version.

closed 2021-01-20 by MaxG87