cilynx/rtl88x2bu #8: Kernel log getting flooded

ajgringo619, 2020-01-23:
[Mint 19.3, kernel 5.3.0-26, driver v5.6.1]

Since upgrading to the v5.6.1 driver (needed to build properly on Mint's latest 5.3 kernel), my kernel log is getting flooded with this (about every 10-15 seconds):

```
Jan 22 22:10:55 kernel: RTW: Turbo EDCA =0xa42b
Jan 22 22:10:57 kernel: RTW: Turbo EDCA =0x5ea42b
```
How can I disable this? The driver seems to be working fine - no issues with disconnections, internet speeds are great.

CGarces, 2020-01-23:
If comes from https://github.com/cilynx/rtl88x2bu/blob/11522ba5fabfa7a60c078258d4161f0b6d21b188/hal/hal_dm.c#L731 so changing the log level should work.

Log level can be changed on run-time so there is no need to compile a new version

Please read [this document](https://github.com/bsdfirst/rtl8814au_EW7833UAC/blob/master/Document/How_to_set_driver_debug_log_level.pdf) for more information

ajgringo619, 2020-01-23:
Thanks a bunch! Changing it to level 2 did the trick.

closed 2020-01-23 by ajgringo619