cilynx/rtl88x2bu #9: New version 5.6.1.6 available

esemeniuc, 2020-01-24:
Download http://www.cudytech.com/wu1200_software_download

Version string:
RTL88x2BU_WiFi_linux_v5.6.1.6_35492.20191025_COEX20180928-6a6a

martadinata666, 2020-02-20:
add usb device list from cillynx usb_intf.c to the new 5.6.1.6 usb_intf.c , success compile at upto kernel 5.5, fail compile in kernel 5.6, in my test.

cilynx, 2020-04-08:
New branch created:
https://github.com/cilynx/rtl88x2bu/tree/5.6.1.6_35492.20191025_COEX20180928-6a6a

Just starting cleanup, but feel free to test.  I confirmed it compiles on 5.4, but it has many of the old usual suspect compiler warnings.

Once we get it cleaned up and have a few users confirm functionality, we'll make that the default branch on the repo.

closed 2020-04-08 by cilynx