cilynx/rtl88x2bu #12: Issue while building dkms

Juliangg1, 2020-02-01:
Hey,
First of all thank you so much for at least keeping my dongle alive.
But so far I tried to follow your installation guide, but it always fails at building+installing the dkms. I am using Kali Linux on 5.4.0 and an tpLink T4U as dongle.
Hope you can help me get started again.

bjlockie, 2020-02-28:
Post the error.

d4rkw1zard, 2020-03-04:
I'm receiving the same error:

`# sudo dkms add -m rtl88x2bu -v ${VER}

Creating symlink /var/lib/dkms/rtl88x2bu/5.6.1/source ->
                 /usr/src/rtl88x2bu-5.6.1

DKMS: add completed.
root@c0st3ll0:/home/majic/Downloads/rtl88x2bu# sudo dkms build -m rtl88x2bu -v ${VER}

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.4.0-kali4-amd64 KVER=5.4.0-kali4-amd64 src=/usr/src/rtl88x2bu-5.6.1....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.0-kali4-amd64 (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.6.1/build/make.log for more information.

DKMS make.log for rtl88x2bu-5.6.1 for kernel 5.4.0-kali4-amd64 (x86_64)
Tue 03 Mar 2020 08:54:41 PM CST
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.4.0-kali4-amd64/build M=/var/lib/dkms/rtl88x2bu/5.6.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-kali4-amd64'
/bin/sh: 1: bc: not found
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_io.o
/usr/src/rtl88x2bu-5.6.1/core/rtw_debug.c: In function ‘dump_drv_version’:
/usr/src/rtl88x2bu-5.6.1/core/rtw_debug.c:45:62: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
   45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |                                                              ^
In file included from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:60,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_debug.c:17:
/usr/src/rtl88x2bu-5.6.1/core/rtw_debug.c:45:44: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
   45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |                                            ^~~~~~~~
/usr/src/rtl88x2bu-5.6.1/include/rtw_debug.h:114:34: note: in definition of macro ‘RTW_PRINT’
  114 |    _dbgdump(DRIVER_PREFIX fmt, ##arg);\
      |                                  ^~~
/usr/src/rtl88x2bu-5.6.1/core/rtw_debug.c:45:2: note: in expansion of macro ‘RTW_PRINT_SEL’
   45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |  ^~~~~~~~~~~~~
/usr/src/rtl88x2bu-5.6.1/core/rtw_debug.c:45:54: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
   45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |                                                      ^~~~~~~~
/usr/src/rtl88x2bu-5.6.1/include/rtw_debug.h:114:34: note: in definition of macro ‘RTW_PRINT’
  114 |    _dbgdump(DRIVER_PREFIX fmt, ##arg);\
      |                                  ^~~
/usr/src/rtl88x2bu-5.6.1/core/rtw_debug.c:45:2: note: in expansion of macro ‘RTW_PRINT_SEL’
   45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |  ^~~~~~~~~~~~~
/usr/src/rtl88x2bu-5.6.1/core/rtw_debug.c:45:44: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
   45 |  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |                                            ^~~~~~~~
/usr/src/rtl88x2bu-5.6.1/include/rtw_debug.h:224:25: note: in definition of macro ‘RTW_PRINT_SEL’
  224 |    _seqdump(sel, fmt, ##arg) /*rtw_warn_on(1)*/; \
      |                         ^~~
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme.o
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-5.4.0-kali4-common/scripts/Makefile.build:271: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_debug.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/usr/src/linux-headers-5.4.0-kali4-common/Makefile:1665: /var/lib/dkms/rtl88x2bu/5.6.1/build] Error 2
make[1]: *** [/usr/src/linux-headers-5.4.0-kali4-common/Makefile:179: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-kali4-amd64'
make: *** [Makefile:2270: modules] Error 2`

d4rkw1zard, 2020-03-04:
I was able to fix this by `apt update` and reinstalling the required packages in apt.

kinsbrunner, 2020-04-10:
I solved this by doing `sudo apt-get install bc`

nizzal, 2020-06-29:
@kinsbrunner thanks it did help me. Thanks a lot! ^_^

closed 2021-01-20 by MaxG87