cilynx/rtl88x2bu #14: Kernel 5.6 RC3: Error! Bad return status for module build on kernel: 5.6.0-050600rc3-generic (x86_64)

TheJags, 2020-02-24:
Building module fails while trying to install Kernel 5.6 RC3 in Ubuntu MATE 20.04 with this error message:

```
Building module:

cleaning build area...

make -j12 KERNELRELEASE=5.6.0-050600rc3-generic KVER=5.6.0-050600rc3-generic src=/usr/src/rtl88x2bu-5.6.1......(bad exit status: 2)

Error! Bad return status for module build on kernel: 5.6.0-050600rc3-generic (x86_64)

Consult /var/lib/dkms/rtl88x2bu/5.6.1/build/make.log for more information.
   ...done.
```

Here's the **/var/lib/dkms/rtl88x2bu/5.6.1/build/make.log**:

```
DKMS make.log for rtl88x2bu-5.6.1 for kernel 5.6.0-050600rc3-generic (x86_64)

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.6.0-050600rc3-generic/build M=/var/lib/dkms/rtl88x2bu/5.6.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.6.0-050600rc3-generic'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_btcoex.o
/usr/src/rtl88x2bu-5.6.1/core/rtw_ap.c: In function ‘rtw_add_bcn_ie’:
/usr/src/rtl88x2bu-5.6.1/core/rtw_ap.c:193:25: warning: ‘ielen’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  193 |  if (p != NULL && ielen > 0) {
      |                   ~~~~~~^~~
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/rtw_proc.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/ioctl_mp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_intf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_com.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_com_phycfg.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_phy.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_dm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_dm_acs.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_btcoex.o
/usr/src/rtl88x2bu-5.6.1/os_dep/linux/rtw_proc.c: In function ‘rtw_proc_create_entry’:
/usr/src/rtl88x2bu-5.6.1/os_dep/linux/rtw_proc.c:70:71: error: passing argument 4 of ‘proc_create_data’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   70 |  entry = proc_create_data(name,  S_IFREG | S_IRUGO | S_IWUGO, parent, fops, data);
      |                                                                       ^~~~
      |                                                                       |
      |                                                                       const struct file_operations *
In file included from /usr/src/rtl88x2bu-5.6.1/include/../os_dep/linux/rtw_proc.h:18,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_intf.h:128,
                 from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:106,
                 from /usr/src/rtl88x2bu-5.6.1/os_dep/linux/rtw_proc.c:17:
./include/linux/proc_fs.h:59:31: note: expected ‘const struct proc_ops *’ but argument is of type ‘const struct file_operations *’
   59 | extern struct proc_dir_entry *proc_create_data(const char *, umode_t,
      |                               ^~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_mp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_mcc.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_hci/hal_usb.o
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:268: /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/rtw_proc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1681: /var/lib/dkms/rtl88x2bu/5.6.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.6.0-050600rc3-generic'
make: *** [Makefile:2270: modules] Error 2
```

Many thanks.

TheJags, 2020-02-28:
I should add the make and model of the adapter:
`TP-Link Archer T3U dual-band USB wireless adapter`

TP-Link: `https://www.tp-link.com/us/support/download/archer-t3u/#Driver`

OS and kernel:
`Ubuntu MATE 19.10 64-bit`
`Ubuntu MATE 20.04 64-bit`
`Kernel version: 5.6 RC3`

tester1969, 2020-03-23:
Created a fork for 5.6 can you try?

https://github.com/tester1969/rtl88x2bu

TheJags, 2020-03-23:
@tester1969 yes, it worked fine with kernel 5.6 RC7.

Though after installing the drivers from your fork, I only used the PC for 5 minutes but the wireless adapter was working just fine. Thanks alot :)

QuercusFelis, 2020-04-07:
I can also confirm that this patch fixes the build issue.

cilynx, 2020-04-08:
Can y'all please confirm if everything is working for you now that #29 has been merged?

tester1969, 2020-04-08:
It compiles but I d rather not try it right now. BTW I have noticed that the driver with iwconfig doesn't give the actual connection speed but the maximum possible. Is this correct?

TheJags, 2020-04-08:
> Can y'all please confirm if everything is working for you now that #29 has been merged?
@cilynx In order to try, I would need to first remove the drivers/modules I installed it from tester1969's fork, right? Thanks.

By the way, I'm running kernel 5.6.2 on Ubuntu MATE 20.04

tester1969, 2020-04-08:
Yes you have to remove mine, find them under /lib/modules, mine are in /lib/modules/5.6.2-300.fc32.x86_64/extra. Of course you must be on the console, because you are going to loose connectivity.

TheJags, 2020-04-08:
@tester1969  @cilynx 

First I tried installing drivers from: `5.6.1.6_35492.20191025_COEX20180928-6a6a`
branch thinking this are the "latest" drivers, but installation failed. I'm gonna post the `make.log` in next post.

Then I tried
`5.6.1_30362.20181109_COEX20180928-6a6a`
branch and installation went through just fine.

Thank you guys :)

cilynx, 2020-04-08:
You are correct that `5.6.1.6` is the latest upstream code.  I've only tested it on Kernel 5.4 though -- what distros are you folks running?  Are you running packaged kernels or compiling your own to stay bleeding edge?

TheJags, 2020-04-08:
`/var/lib/dkms/rtl88x2bu/5.6.1/build/make.log`
from a failed installation for
`5.6.1.6_35492.20191025_COEX20180928-6a6a` branch (as mentioned in the earlier post):

````
DKMS make.log for rtl88x2bu-5.6.1 for kernel 5.6.3-050603-generic (x86_64)

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.6.3-050603-generic/build M=/var/lib/dkms/rtl88x2bu/5.6.1/build  modules

make[1]: Entering directory '/usr/src/linux-headers-5.6.3-050603-generic'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/rtw_proc.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/ioctl_mp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_intf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_com.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_com_phycfg.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_phy.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_dm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_dm_acs.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_btcoex.o

/usr/src/rtl88x2bu-5.6.1/os_dep/linux/rtw_proc.c: In function ‘rtw_proc_create_entry’:
/usr/src/rtl88x2bu-5.6.1/os_dep/linux/rtw_proc.c:70:71: error: passing argument 4 of ‘proc_create_data’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   70 |  entry = proc_create_data(name,  S_IFREG | S_IRUGO | S_IWUGO, parent, fops, data);
      |                                                                       ^~~~
      |                                                                       |
      |                                                                       const struct file_operations *
In file included from /usr/src/rtl88x2bu-5.6.1/include/../os_dep/linux/rtw_proc.h:18,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_intf.h:128,
                 from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:106,
                 from /usr/src/rtl88x2bu-5.6.1/os_dep/linux/rtw_proc.c:17:
./include/linux/proc_fs.h:59:31: note: expected ‘const struct proc_ops *’ but argument is of type ‘const struct file_operations *’
   59 | extern struct proc_dir_entry *proc_create_data(const char *, umode_t,
      |                               ^~~~~~~~~~~~~~~~
[... cut ...]
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/hal_hci/hal_usb.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/hal/led/hal_led.o
cc1: some warnings being treated as errors

make[2]: *** [scripts/Makefile.build:268: /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/rtw_proc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1683: /var/lib/dkms/rtl88x2bu/5.6.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.6.3-050603-generic'
make: *** [Makefile:2283: modules] Error 2
````

TheJags, 2020-04-08:
@cilynx I have just posted the error log for `5.6.1.6_35492.20191025_COEX20180928-6a6a` branch.

I'm running kernel 5.6.3 on Ubuntu MATE 20.04 64-bit.

TheJags, 2020-04-08:
@cilynx i'm running mainline kernel 5.6.3, installed from this page:
`https://kernel.ubuntu.com/~kernel-ppa/mainline/`
Thanks alot.

tester1969, 2020-04-08:
This is the known error we fixed. I dunno if it should exist in this branch. Anyway, I am running fedora, because it must be somehow close to my work laptop which is RHEL. I am running beta fc32 with updates-testing, although I am not compiling I am to the latest. 
I started my career as system assembly etc programmer and penetration tester so I was very much into programming but nowayadys work is far from this.. you know damn managerial. 
I would gladly test anything knew, however.

cilynx, 2020-04-08:
Thanks for the error log @TheJags -- As @tester1969 mentioned that's the same problem addressed in #29 on the `5.6.1` branch.  I need to refactor @karubabu's patch for `5.6.1.6`.  I should be able to hit this in the next couple days if no one get to it before me.

TheJags, 2020-04-08:
> Thanks for the error log @TheJags -- As @tester1969 mentioned that's the same problem addressed in #29 on the `5.6.1` branch. I need to refactor @karubabu's patch for `5.6.1.6`. I should be able to hit this in the next couple days if no one get to it before me.

Alright, I will try again once you update the `5.6.1.6_35492.20191025_COEX20180928-6a6a` branch.

Thank you so much :)

mattiasmun, 2020-05-11:
Hello.

I am getting a new build error for branch 5.6.1_30362.20181109_COEX20180928-6a6a.

Can you please check it out?

DKMS make.log for rtl88x2bu-5.6.1 for kernel 5.6.10-300.fc32.x86_64 (x86_64)
monday 11 may 2020 09:06:40 CEST
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.6.10-300.fc32.x86_64/build M=/var/lib/dkms/rtl88x2bu/5.6.1/build  modules
make[1]: Entering directory ”/usr/src/kernels/5.6.10-300.fc32.x86_64”

In file: os_dep/linux/ioctl_mp.c

”tree_vec” not supported by expression””
during the GIMPLE-pass: strlen
In function ”rtw_mp_txpower_index”:
Segmentation fault
  544 | int rtw_mp_txpower_index(struct net_device *dev,
      |     ^~~~~~~~~~~~~~~~~~~~

make[2]: *** [scripts/Makefile.build:268: /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/ioctl_mp.o] Error 1
make[2]: *** Waiting for unfinished jobs...
make[1]: *** [Makefile:1683: /var/lib/dkms/rtl88x2bu/5.6.1/build] Error 2
make[1]: Leaving directory ”/usr/src/kernels/5.6.10-300.fc32.x86_64”
make: *** [Makefile:2283: modules] Error 2

silvernode, 2020-05-13:
I am trying not to open a new issue so I will post my make log here and it can be moved to a separate issue if need be. 

So long story short, I have been using the rtl88x2bu dkms module since kernel 5.3 and it continued to work fine all the way through kernel 5.6. I can always rebuild the module manually but the build always fails when dkms hooks to a kernel update. This issue has carried over from Fedora 31 to Fedora 32. I have been patient up until now since it does not seem as if this issue has been known/fixed. So here is a gist of my make log that was generated about an hour ago during a kernel update.

https://gist.github.com/silvernode/74d99d1d6b0482500fbe54423dd606e4

TheJags, 2020-06-26:
Closing this one as I'm gonna open a new issue for kernel 5.8 RC2 with similar errors.

closed 2020-06-26 by TheJags

LeoHeinemann, 2020-10-04:
It worked for me as well.
Could run my Aoyool AC1200 WLAN Usb3.0 Adapter with your Fork. Thx, tester1969!