cilynx/rtl88x2bu #18: anybody know how to lessen the RTW: debug messages?

bjlockie, 2020-02-28:
Does anybody know how to lessen the RTW: debug messages?
I tried modifying the Make file and changing:
CONFIG_RTW_DEBUG = n
# default log level is _DRV_INFO_ = 4,
# please refer to "How_to_set_driver_debug_log_level.doc" to set the available level.
I can't find How_to_set_driver_debug_log_level.doc

TheJags, 2020-02-28:
Here's the link to the How_to_set_driver_debug_log_level.pdf:

`https://github.com/bsdfirst/rtl8814au_EW7833UAC/blob/master/Document/How_to_set_driver_debug_log_level.pdf`

The command you would need to run is (or change "echo 2" to the desired one):
`echo 2 > /proc/net/rtl88x2bu/log_level`

 and the command to find the current log level is:
`cat /proc/net/rtl88x2bu/log_level`

bjlockie, 2020-02-29:
```
$ cat /proc/net/rtl88x2bu/log_level
drv_log_level:4
  _DRV_NONE_ = 0 = 0
  _DRV_ALWAYS_ = 1 = 1
  _DRV_ERR_ = 2 = 2
  _DRV_WARNING_ = 3 = 3
+ _DRV_INFO_ = 4 = 4
  _DRV_DEBUG_ = 5 = 5
  _DRV_MAX_ = 6 = 6

```

bjlockie, 2020-02-29:
I've probably already screwed up my copy of the repo. :-)

bjlockie, 2020-02-29:
wrong module name :-)

closed 2020-02-29 by bjlockie