cilynx/rtl88x2bu #19: Does anyone see "IEEE 802.11AC"?

bjlockie, 2020-02-29:
Does anyone see "IEEE 802.11AC"?

$ iwconfig
wlx1cbfce9e4193  **IEEE 802.11a**  ESSID:

TheJags, 2020-02-29:
I do get to see IEEE 802.11AC if I set the router to 5.0 GHz. My router have dual bands but it's not simultaneous, so it's either 5.0 GHz or 2.4 GHz.

`iwconfig` results:
```
When router is set to 5.0 GHz:
wlxd0374547816a  IEEE 802.11AC  ESSID:"SSID-Name_optout_nomap"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:5.2 GHz  Access Point: <MAC ID>
          Bit Rate:434 Mb/s   Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=94/100  Signal level=-34 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

When router is set to 2.4 GHz:
wlxd0374547816a  IEEE 802.11bgn  ESSID:"SSID-Name_optout_nomap"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.412 GHz  Access Point: <MAC ID>
          Bit Rate:200 Mb/s   Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=87/100  Signal level=-49 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0```
```

bjlockie, 2020-02-29:
Does your device show up as USB3?
lsusb -v -d 0bda:b812 probably (lsusb will show you the device (eg. 0bda:b812)).
Look for 'bcdUSB               xxxx' and
'Device can operate at Full Speed (12Mbps)
Device can operate at High Speed (480Mbps)'

TheJags, 2020-02-29:
This TP-Link Archer T3U wireless adapter is connected to the PC via **USB 2.0 port**. This desktop does not have USB 3.0 port.

Here's the result of `lsusb -v -d 2357:012d` anyway:
```
Bus 001 Device 003: ID 2357:012d TP-Link 802.11ac NIC
Device Descriptor:
  bcdUSB               2.10
  idVendor           0x2357 TP-Link
  bcdDevice            2.10
  iManufacturer           1 Realtek
  iProduct                2 802.11ac NIC
... ... ...
Binary Object Store Descriptor:
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      HIRD Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x0006
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
```

bjlockie, 2020-02-29:
This might be telling "SuperSpeed USB Device Capability".
I suspect the interface is USB 3 but the internals are USB 2.
Mine won't connect at USB 3 even in a USB 3 port,
It says the same as yours.
I'll ask the USB people. :-)
Thanks.

bjlockie, 2020-03-02:
The module has a parameter to switch to USB 3 mode.
Put
options 88x2bu rtw_switch_usb_mode=1
in /etc/modeprobe.d/88x2bu.conf on ubuntu.
It seems "fake" (maybe fooling the USB subsystem.
I don't think the driver has full 80211AC support for use as an AP so I've ordered a 8812AU based chip (a bit older but I'm hoping to use it as an access point).