cilynx/rtl88x2bu #21: This version has unstable 5GHz connection

fantasist, 2020-03-04:
I tried to watch a 1080p60fps stream on Twitch. Video frequently freezes on this version for unknown reason. I didn't have this issue on windows 10 and linux with driver version 5.3.1 (which coincidentally is the version coming with driver CD). I manually built 5.6.1.5 from source, replaced the .ko, then the connection was stable again. I think other people might experience the same issue and it's probably worth upgrading, thanks!

QuercusFelis, 2020-04-07:
I can confirm my connection seems to fail randomly when using 5Ghz

tvforthedemon, 2021-07-08:
I can't connect to a 5G network at all with branch 5.8.7.1_35809.20191129_COEX20191120-7777. Going to uninstall and go with 5.6.1_30362.20181109_COEX20180928-6a6a. If that doesn't work. I'll give what fantasist said a shot. IF that doesn't work I guess ill go back to low signal strength or get a properly supported card