cilynx/rtl88x2bu #22: Installed. No errors. No device found.

DanielTheSilly, 2020-03-20:
I don't really know what I'm doing here and need some help. I'm not 100% sure I installed this right, but I followed the instructions and a
`dkms status`
Gives me
`nvidia, 435.21, 4.15.0-48-generic, x86_64: installed
nvidia, 435.21, 4.15.0-91-generic, x86_64: installed
rtl88x2bu, 5.6.1, 4.15.0-48-generic, x86_64: installed
rtl88x2bu, 5.6.1, 4.15.0-91-generic, x86_64: installed`
If I run
`iwconfig`
I get
`lo        no wireless extensions.

enp4s0    no wireless extensions.`
The dongle is pluged in and listed with
`lsusb`
here
`Bus 003 Device 005: ID 0846:9055 NetGear, Inc. `

This is basically all I've done. From what I can see it's installed so it looks like I just have to get some other program to recognize it, but I don't know how. I thought it would just pop up in iwconfig or something.

DanielTheSilly, 2020-04-01:
Ok. After all this time I finally found the issue and managed to get it working. rtl88x2bu/os_dep/linux/usb_intf.c is missing a listing for the device. I had to read a lot of the code to figure out that realtek doesn't have an actual drivers for RTL8812BU uses RTL8822B as the driver and just detects differences. Wacky... But whatever. I'll submit a fix or clone the repo if it isn't accepted. Please close this issue once the fix is implemented. Thanks.

Daquartzinator, 2020-04-14:
I'm having a very similar error. My module is installed and active, but the device isn't running the drivers.

I started on version 5.2.4.4, because that was the version of the drivers my device came with. I then tried updating to 5.6.1. I've gotten almost the exact same things you were getting earlier:

dkms status gives
rtl88x2bu, 5.2.4.4, 4.15.0-96-generic, i686: built
rtl88x2bu, 5.6.1, 4.15.0-96-generic, i686: installed

and a lsusb shows: Bus 001 Device 005: ID 0bda:c811 Realtek Semiconductor Corp.

I added the values from lsusb to the usb_intf.c file, but I'm not sure if I rebuilt the module correctly (I'm a new linux user). It still didn't work after that. How did you get it to work for you?

DanielTheSilly, 2020-04-15:
Hey. Looks like you are right and it has the same problem. I found someone that cracked his open and checked that it's a RTL8811CU on that device ID and from someone ELSE that the vendor ID is 0x0bda. I'll upload a fix for you, but do keep in mind that given that I don't have the device I have no way to test or check the information to see if this is right and messing with drivers this way is not really recommended. If you want to be safe you should confirm everything I said is right before you try it out. Hope that helps.

Daquartzinator, 2020-04-16:
Thank you for your help. I was actually able to get it working: I found the vendor and product ID from doing an lsusb, which wound up being the only way I was able to identify the device. When I looked it up, I found a page of people trying to set the drivers up, and one person recommended this github repo: https://github.com/brektrou/rtl8821CU.git
So it turns out that it runs the 8821CU version of the drivers.
(I checked back in the file of supported devices under the 8821CU, and sure enough, it was there.)
Thank you for clearing that stuff up, the issue I had before makes sense now.

jrchavesSA, 2020-04-16:
I am trying to get the wifi dongle into monitor mode for the use of my course. Can u help? -- Best Regards,  CREATIVE SPECIALIST Cell: ‪084-482 8527‬Website: www.jrchaves.co.za-------- Original message --------From: Quartz <notifications@github.com>Date: Thu, Apr 16, 2020, 22:49To: cilynx/rtl88x2bu <rtl88x2bu@noreply.github.com>Cc: Subscribed <subscribed@noreply.github.com>Subject: Re: [cilynx/rtl88x2bu] Installed. No errors. No device found. (#22)
Thank you for your help. I was actually able to get it working: I found the vendor and product ID from doing an lsusb, which wound up being the only way I was able to identify the device. When I looked it up, I found a page of people trying to set the drivers up, and one person recommended this github repo: https://github.com/brektrou/rtl8821CU.git
So it turns out that it runs the 8821CU version of the drivers.
(I checked back in the file of supported devices under the 8821CU, and sure enough, it was there.)
Thank you for clearing that stuff up, the issue I had before makes sense now.

closed 2020-04-21 by DanielTheSilly