cilynx/rtl88x2bu #26: Issue compiling on a raspberry pi 3b+

massimilianonball, 2020-03-29:
Hi all!!
My system based on raspbian buster:
`uname -a
Linux RaspFly 4.19.108-v7+ #1298 SMP Fri Mar 6 18:04:35 GMT 2020 armv7l GNU/Linux`

I following the procedure of installation suggested on the README page:
```
# Pull down the driver source
git clone https://github.com/cilynx/rtl88x2bu
cd rtl88x2bu/

# Configure for RasPi
sed -i 's/I386_PC = y/I386_PC = n/' Makefile
sed -i 's/ARM_RPI = n/ARM_RPI = y/' Makefile

# DKMS as above
VER=$(sed -n 's/\PACKAGE_VERSION="\(.*\)"/\1/p' dkms.conf)
sudo rsync -rvhP ./ /usr/src/rtl88x2bu-${VER}
sudo dkms add -m rtl88x2bu -v ${VER}
sudo dkms build -m rtl88x2bu -v ${VER} # Takes ~3-minutes on a 3B+
Error! echo
Your kernel headers for kernel 4.19.108-v7+ cannot be found at
/lib/modules/4.19.108-v7+/build or /lib/modules/4.19.108-v7+/source.
```
Any idea to solve the problem and complete the compiling process?
Thnaks!!

cilynx, 2020-03-29:
You need to install `linux-headers` for your installed kernel in order to compile modules.  `apt-cache search linux-headers` will show you the available versions.

massimilianonball, 2020-03-29:
Many thanks for the quick answer!!

```
uname -r
4.19.108-v7+
apt-cache search linux-headers-4.*
linux-headers-4.18.0-3-common - Common header files for Linux 4.18.0-3
linux-headers-4.18.0-3-common-rt - Common header files for Linux 4.18.0-3-rt
linux-headers-4.9.0-6-all - All header files for Linux 4.9 (meta-package)
linux-headers-4.9.0-6-all-armhf - All header files for Linux 4.9 (meta-package)
linux-headers-4.9.0-6-common - Common header files for Linux 4.9.0-6
linux-headers-4.9.0-6-common-rt - Common header files for Linux 4.9.0-6-rt
linux-headers-4.9.0-6-rpi - Header files for Linux 4.9.0-6-rpi
linux-headers-4.9.0-6-rpi2 - Header files for Linux 4.9.0-6-rpi2
```

At the moment they are not available.. I must wait for the 4.19 version or i can try to compile with the 4.18.0-3?

cilynx, 2020-03-29:
In order for the module to work, the headers you compile against must match your running kernel exactly.  I forgo that RasPi has its own header package.  See here:

https://www.raspberrypi.org/documentation/linux/kernel/headers.md

massimilianonball, 2020-03-30:
So if i have understand in the right way, i can use the raspberry pi kernel header instead of the generic kernel, right?
This way haven't particular downside? Maybe i should recompile the driver at every kernel change?
Many thanks for your informations

massimilianonball, 2020-03-30:
No luck!!
Even installing the "raspberrypi-kernel-headers" the compiling process fail in the same point with the same error, i report below the terminal output:
```
sudo dkms add -m rtl88x2bu -v ${VER}
Creating symlink /var/lib/dkms/rtl88x2bu/5.6.1/source ->
                 /usr/src/rtl88x2bu-5.6.1

DKMS: add completed.

```
```
sudo dkms build -m rtl88x2bu -v ${VER}
Error! echo
Your kernel headers for kernel 4.19.108-v7+ cannot be found at
/lib/modules/4.19.108-v7+/build or /lib/modules/4.19.108-v7+/source.

```
```
ls /lib/modules
4.19.108+  4.19.108-v7+  4.19.97+  4.19.97-v7+  4.19.97-v7l+  4.19.97-v8+

```
```
ls /lib/modules/4.19.108-v7+/
kernel         modules.alias.bin  modules.builtin.bin  modules.dep.bin  modules.order    modules.symbols
modules.alias  modules.builtin    modules.dep          modules.devname  modules.softdep  modules.symbols.bin

```
I have a little doubt, in the compiling process i have:
```
echo ${VER}
5.6.1
```

${VER} shoulden't be equal to my kernel version (4.19.108-v7+)??

massimilianonball, 2020-03-31:
Any update guys??

cilynx, 2020-03-31:
Can you show me the output from `dpkg -L raspberrypi-kernel-headers`?  `/lib/modules/$KVER/build` should be a part of that package.  On stretch, I see:
```
pi@piab:~ $ dpkg -L raspberrypi-kernel-headers | grep modules.*build
/lib/modules/4.19.66+/build
/lib/modules/4.19.66-v7+/build
pi@piab:~ $
```

massimilianonball, 2020-03-31:
There it is:

```
dpkg -L raspberrypi-kernel-headers | grep modules.*build
/lib/modules/4.19.97+/build
/lib/modules/4.19.97-v7+/build
/lib/modules/4.19.97-v7l+/build
```
Other informations:
```
uname -a
Linux RaspFly 4.19.108-v7+ #1298 SMP Fri Mar 6 18:04:35 GMT 2020 armv7l GNU/Linux
cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
ls /lib/modules/
4.19.108+  4.19.108-v7+  4.19.97+  4.19.97-v7+  4.19.97-v7l+  4.19.97-v8+
```
I've recently update from raspbian stretch to buster.

cilynx, 2020-04-01:
There's your issues -- your running kernel is `4.19.108-v7+`, but your installed kernel headers are `4.19.97-v7+`.  You need to either upgrade you kernel headers or downgrade your running kernel so that they're the same version.

massimilianonball, 2020-04-01:
Ok got it, this morning i've tried to give a:

`apt search linux-headers-$(uname -r)`

No results, so i think that i will try to downgrade the kernel.

massimilianonball, 2020-04-01:
Problem solved!! Many thanks for your help!!
I have downgraded the kernel to the last stable version:
`sudo apt-get update && sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel`
I have found the cause of the problem of different kernel/headers versions: upgrading from stretch to buster i have used an "rpi-update" too.. Reading the raspbian documentation i have found:

> "In normal circumstances there is NEVER a need to run rpi-update as it always gets you to the leading edge firmware and kernel and because that may be a testing version it could leave your RPi unbootable". https://www.raspberrypi.org/forums/viewtopic.php?p=916911#p916911 Even the rpi-update documentation now warns "Even on Raspbian you should only use this with a good reason. This gets you the latest bleeding edge kernel/firmware."

Again many thanks for your help!!

cilynx, 2020-04-08:
Glad you got it sorted -- cheers!

closed 2020-04-08 by cilynx