cilynx/rtl88x2bu #28: Issue with TP Link AC1300 T3U(EU) ver. 1 on Ubuntu 18.04

robpal1990, 2020-03-30:
Hello,

I have successfully compiled and installed the driver (I have tried earlier with 8822bu drivers by jeremy31 with no success, unistalled them).

The internet connection worked for the day and after a reboot I cannot make it work again. On the other hand, the command

```lsusb``` returns for ```Bus 001 Device 002: ID 2357:012d``` with no further description. The description was absent even when the connection worked fine earlier today.

The adapter sees the wireless networks but when I try to connect it never happens. I tried recompiling with older versions of the driver but the situation was the same.

My kernel vesion is ```5.3.0-28-generic```.

Any ideas how I could make it work?

Nic69Han, 2020-05-22:
Hello,
I have the same problem like @robpal1990 , only it works fine with kernel 5.0.0-48 but not with any 5.3 kernel.
I don't have a usb3 port to test. 
I haven't found a solution yet.

Nic69Han, 2020-05-31:
Hello,
It works very well with the 5.4.0-33 kernel. (Ubuntu 20.04)
thanks

straaljager, 2020-06-02:
Same here, this driver works great with TP-Link Archer AC1300 T4U ver 3 on Ubuntu 20.04 (kernel 5.4.0-33 generic)

pitrou, 2022-04-15:
Just for the record and unlike what's mentioned above, the driver works very badly for me with TP-LINK Archer T4U v3 on Ubuntu 20.04 (kernel 5.4.0-107-generic). Issue reported at https://github.com/cilynx/rtl88x2bu/issues/203