cilynx/rtl88x2bu #32: 4-way handshake fails most of the time (Asus USB-AC53 Nano)

nmggithub, 2020-04-16:
Ok, so I originally tried a driver for my device and it worked very well. Then it stopped working, so I looked for another driver. I came across several, but I am using this as it appears to be the best maintained. When connecting to password protected WiFi networks, the 4-way handshake ends up failing as messages 1 and 2 simply repeat back and forth until there is a timeout. This is confirmed via Wireshark.

tester1969, 2020-04-24:
This seems the old bug of not having the correct driver/kernel. What kernel are you?
I have the same asus dongle and works perfect

nmggithub, 2020-04-25:
I am on kernel 4.19.0-8-amd64. What version should I be on?

tester1969, 2020-04-25:
Hmm I don't know how to answer that. Is there a special reason not to go to kernel 5.6.x ? What flavor do you have?

nmggithub, 2020-04-25:
I'm on Debian,and going up to 5.6 would require going on an unstable branch. I think I'm going to try it though.

nmggithub, 2020-04-25:
Wait, it appears Debian doesn't even support that yet. I might go with Arch.