cilynx/rtl88x2bu #37: Having trouble making this work.

IrnBro, 2020-04-26:
I'm running Xubuntu 20.04, Linux version is 5.4.0-26.

Adapter in question is a TP-Link AC1300 v3.

I pretty much copied and pasted all the instructions from the DKMS section into my terminal, and they all seemed to complete.

The adapter shows up in lsusb but it doesnt look like the computer knows what to do with it.

Any help would be greatly appreciated.

MarcCoru, 2020-04-26:
I have a similar issue on a clean Ubuntu 20.04 with linux kernel `Linux 5.4.0-26-generic`.
I use the Fideco USB 3.0 Dual Band Wireless USB adapter AC-1200Mbps ([amazon link](https://www.amazon.de/-/en/gp/product/B08523KPP9/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&psc=1))
installation (from readme) completed without errors, but the device is not recognized.

The `dmesg` after plugging it in shows:
```
[  212.503795] Workqueue: usb_hub_wq hub_event
[  212.503818] RIP: 0010:rtw_os_ndevs_register+0xb6/0xde [88x2bu]
[  212.503819] Code: 87 d4 47 00 00 49 8d b6 bb 33 00 00 84 c0 74 0d 49 8d b6 cb 33 00 00 3c 01 49 0f 45 f7 e8 0c f3 ff ff 41 89 c5 83 f8 01 74 b0 <0f> 0b 85 db 75 15 48 63 db 49 8b bc dc 50 01 00 00 48 85 ff 74 05
[  212.503819] RSP: 0018:ffffa75bc03577b0 EFLAGS: 00010297
[  212.503820] RAX: 0000000000000000 RBX: 0000000000000000 RCX: ffff964d038aa060
[  212.503820] RDX: ffff964d038aa060 RSI: 0000000000000007 RDI: ffffa75bc02fe548
[  212.503821] RBP: ffffa75bc03577e0 R08: 0000000000000000 R09: 0000000000000006
[  212.503821] R10: ffff964d0ebe9a00 R11: ffff964d0ebe99e0 R12: ffff964d038a9000
[  212.503822] R13: 0000000000000000 R14: ffffa75bc02fb000 R15: ffffffffc0862e05
[  212.503822] FS:  0000000000000000(0000) GS:ffff964d0ebc0000(0000) knlGS:0000000000000000
[  212.503823] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  212.503823] CR2: 00007f085c3779c0 CR3: 000000021dc0a005 CR4: 00000000001606e0
[  212.503824] Call Trace:
[  212.503846]  rtw_os_ndevs_init+0x29/0x3d [88x2bu]
[  212.503868]  rtw_drv_init+0x2a8/0x35a [88x2bu]
[  212.503869]  usb_probe_interface+0xed/0x290
[  212.503870]  really_probe+0x2b3/0x3e0
[  212.503871]  driver_probe_device+0xbc/0x100
[  212.503872]  __device_attach_driver+0x71/0xd0
[  212.503873]  ? driver_allows_async_probing+0x50/0x50
[  212.503874]  bus_for_each_drv+0x84/0xd0
[  212.503875]  __device_attach+0xe1/0x160
[  212.503876]  device_initial_probe+0x13/0x20
[  212.503877]  bus_probe_device+0x8f/0xa0
[  212.503878]  device_add+0x3c7/0x6b0
[  212.503879]  ? _cond_resched+0x19/0x30
[  212.503880]  usb_set_configuration+0x433/0x7d0
[  212.503881]  ? kernfs_activate+0x77/0x80
[  212.503882]  generic_probe+0x43/0x60
[  212.503883]  usb_probe_device+0x31/0x70
[  212.503884]  really_probe+0x2b3/0x3e0
[  212.503885]  driver_probe_device+0xbc/0x100
[  212.503886]  __device_attach_driver+0x71/0xd0
[  212.503887]  ? driver_allows_async_probing+0x50/0x50
[  212.503888]  bus_for_each_drv+0x84/0xd0
[  212.503888]  __device_attach+0xe1/0x160
[  212.503889]  device_initial_probe+0x13/0x20
[  212.503890]  bus_probe_device+0x8f/0xa0
[  212.503891]  device_add+0x3c7/0x6b0
[  212.503893]  ? crng_fast_load+0x110/0x140
[  212.503894]  usb_new_device.cold+0x12d/0x304
[  212.503895]  hub_port_connect+0x500/0x850
[  212.503896]  port_event+0x502/0x780
[  212.503897]  ? __switch_to_asm+0x34/0x70
[  212.503899]  hub_event+0x152/0x390
[  212.503900]  ? psi_avgs_work+0x64/0xd0
[  212.503901]  process_one_work+0x1eb/0x3b0
[  212.503902]  worker_thread+0x4d/0x400
[  212.503903]  kthread+0x104/0x140
[  212.503904]  ? process_one_work+0x3b0/0x3b0
[  212.503905]  ? kthread_park+0x90/0x90
[  212.503906]  ret_from_fork+0x35/0x40
[  212.503907] ---[ end trace 702112e3d3767092 ]---
[  212.631939] usb 3-10: reset high-speed USB device number 7 using xhci_hcd
```

MarcCoru, 2020-05-02:
I ran `apt update` and `apt upgrade` which updated the Linux Kernel from 5.4.0-26-generic to 5.4.0-28-generic.
After reboot (on the new kernel) and `sudo dkms install rtl88x2bu/5.6.1` the Wifi works now on my Ubuntu 20.04 machine. @IrnBro maybe that will fix your issue as well.