cilynx/rtl88x2bu #39: 5ghz monitor not work, how to uninstalll driver?

sagaardu, 2020-04-28:
Hello,

After installing the driver the monitor mode does not work in the 5ghz network.

How can I remove everything and remove everything that has been executed?

Cheers

dandar, 2020-08-09:
Why doesn't anyone answer this question about the monitor mode?  Monitor mode is important for monitoring and security debugging, no?  Does the "fastoe" version work, maybe?  https://github.com/fastoe/RTL8812BU

cilynx, 2020-08-10:
Because everyone who contributes to this repo is a volunteer with many other commitments and I don't think any of us use monitor mode.  You're welcome to check out other versions and I'm sure the community would appreciate if you share your results.  Finding something that works is the first step to porting it into this version.