cilynx/rtl88x2bu #42: compile error

ilippert, 2020-05-09:
Hello,

I had this driver working some weeks ago, with DKMS working. After some weeks of perfect harmony, the DKMS wouldn't build anymore.

now I am trying to reinstall/reactivate the driver, trying anew.

Yet `make` throws this error
```
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.6.8-200.fc31.x86_64/build M=/home/ilippert/Techy/wifi criver ac1200 (rtlb812)/rtl88x2bu  modules
/bin/sh: -c: line 0: syntax error near unexpected token `('
/bin/sh: -c: line 0: `make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.6.8-200.fc31.x86_64/build M=/home/ilippert/Techy/wifi criver ac1200 (rtlb812)/rtl88x2bu  modules'
make: *** [Makefile:2283: modules] Error 1
```

cilynx, 2020-05-09:
It doesn't like the parentheses in your path.  I'd bet when it worked
before, you had it in a different directory, no?  Rename the parent
directory to not contain parentheses and you should be good to go.

ilippert, 2020-05-09:
thanks a lot, indeed, "make" works now.

closed 2020-05-09 by ilippert