cilynx/rtl88x2bu #43: Doesn't work with kernel 5.6.11-arch1-1

BlokenClockCrisps, 2020-05-12:
I have compiled the module with these options:
`CONFIG_RTW_DEBUG = y
CONFIG_RTW_LOG_LEVEL = 2`

My card TP-Link Archer T4Uv3 fails to connect to the AP almost all the time, having mac randomization in NetworkManager turned off and tried both with wpa_supplicant and iwd, asks for password in an infinite loop and rarely succeeds with connecting.

BlokenClockCrisps, 2020-05-12:
dmesg output:
```
[May13 00:15] RTW: module init start
[  +0.000003] RTW: rtl88x2bu v5.6.1_30362.20181109_COEX20180928-6a6a
[  +0.000002] RTW: build time: May 13 2020 00:11:15
[  +0.000001] RTW: rtl88x2bu BT-Coex version = COEX20180928-6a6a
[  +0.000086] RTW: [HALMAC]11692M
              HALMAC_MAJOR_VER = 1
              HALMAC_PROTOTYPE_VER = 5
              HALMAC_MINOR_VER = 14
              HALMAC_PATCH_VER = 15
[  +0.044075] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ +12.313897] RTW: HW EFUSE
[  +0.000005] RTW: 0x000: 29 81 00 3C  09 00 A1 00  B6 04 64 10  00 00 A3 00  
[  +0.000008] RTW: 0x010: 33 34 34 35  35 35 2E 2E  2E 2E 2E 13  EE EE FF FF  
[  +0.000007] RTW: 0x020: FF FF 1E 1D  1B 1B 19 17  16 14 14 13  1B 1A 1A 1A  
[  +0.000006] RTW: 0x030: 13 EE FF FF  00 FF 0C E0  FF FF 30 30  30 30 30 30  
[  +0.000006] RTW: 0x040: 2C 2C 2C 2C  2C 13 EE 00  FF FF FF FF  1F 1D 1A 1A  
[  +0.000007] RTW: 0x050: 17 16 14 13  12 10 18 16  16 16 13 EE  FF FF 00 FF  
[  +0.000006] RTW: 0x060: 0C E0 FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000007] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000007] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  A6 1B 1A 00  FF FF FF FF  
[  +0.000006] RTW: 0x0C0: FF 09 00 11  00 00 00 00  00 FF 02 FF  FF FF FF FF  
[  +0.000006] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000007] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x100: 57 23 15 01  C1 67 02 D0  37 45 84 AA  DB 09 03 52  
[  +0.000006] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63  
[  +0.000007] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF  
[  +0.000006] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x140: 33 0A 1F 01  00 00 21 0F  FF FF FF FF  FF FF FF FF  
[  +0.000007] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000007] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000007] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x200: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x210: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000007] RTW: 0x220: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x230: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x240: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x250: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x260: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000007] RTW: 0x270: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x280: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x290: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x2A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000007] RTW: 0x2B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x2C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x2D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000006] RTW: 0x2E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000007] RTW: 0x2F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  +0.000021] RTW: hal_com_config_channel_plan chplan:0x26
[  +0.000005] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  +0.336706] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[  +2.502401] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  +0.006010] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  +0.006004] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  +0.009008] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[  +0.000479] RTW: rtw_ndev_init(wlan0) if1 mac_addr=d0:37:45:84:aa:db
[  +0.000535] usbcore: registered new interface driver rtl88x2bu
[  +0.000003] RTW: module init ret=0
[  +0.330980] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[ +12.454929] RTW: wlan0- hw port(0) mac_addr =d0:37:45:84:aa:db
[  +0.673036] RTW: module exit start
[  +0.000001] usbcore: deregistering interface driver rtl88x2bu
[  +0.099113] RTW: rtw_cmd_thread: DriverStopped(True) SurpriseRemoved(False) break at line 576
[  +0.046615] RTW: rtw_ndev_uninit(wlan0) if1
[  +0.078368] RTW: rtw_dev_unload: driver not in IPS
[  +0.379715] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
[  +0.195081] RTW: module exit success

```

BlokenClockCrisps, 2020-05-18:
Problems with connection have resolved when I got rid of NetworkManager, but now my VFIO setup doesn't work at all and I can't connect with bare iwd at all after restart, but I guess this may be a separate issue.

deathwebo, 2020-06-19:
Hello @BlokenClockCrisps I have that exact same usb WiFi dongle and recently I installed Pop!_os  20.04 which comes with kernel 5.6. After installing this drivers the WiFi worked but after a restart i wasn’t able to connect to the networks in range. It some times works on some restarts but most of the time is failing.

Did you manage to solve it ? You mentioned that you removed the network manager and that helped , can you share the steps you followed ? Thanks !

BlokenClockCrisps, 2020-07-16:
> Hello @BlokenClockCrisps I have that exact same usb WiFi dongle and recently I installed Pop!_os 20.04 which comes with kernel 5.6. After installing this drivers the WiFi worked but after a restart [...]

Sorry for the late reply. I just use iwd which can be used as a standalone network manager for wi-fi adapters and I've disabled NetworkManager which seemingly made trouble with the combo. Since I use Arch Linux, your best bet is to refer to their wiki.