cilynx/rtl88x2bu #52: Add packege to Openwrt

Menion2k, 2020-06-09:
Hi

Openwrt snapshot moved to kernel 5.4

Can you work on adding this driver as Openwrt package?

Bye

MaxG87, 2020-12-25:
I don't know about @mabrand , but providing an OpenWRT package is beyond my capabilities. I do not even use OpenWRT, so I could not test anything.

However, I would love to move the project towards better OpenWRT compatibility. Would you be able to provide pull requests?

rootalc, 2021-03-17:
[https://gitlab.com/_jason/openwrt-rtl8812bu-package](https://gitlab.com/_jason/openwrt-rtl8812bu-package)

rootalc, 2021-03-17:
just tested it.
AP mode - working
STA - not tested
iw dev wlan0 scan - working

Found problem: OpenWRT executing _iw dev wlan0 del_ after enabling wifi, causing freeze
so i created issue: [https://github.com/cilynx/rtl88x2bu/issues/141](https://github.com/cilynx/rtl88x2bu/issues/141)

how to start it:
compile and install kmod-rtl88x2bu
execute: ip link set wlan0 up
and you need to start hostapd manually (if you want AP mode)

Boos4721, 2022-03-26:
> just tested it on OpenWRT (master snapshot (OpenWrt SNAPSHOT r16195-8391781389), arch - x86_64). AP mode - working STA - not tested iw dev wlan0 scan - working
> 
> Found problem: OpenWRT executing `iw dev wlan0 del` after enabling wifi, causing freeze so i created issue: #141
> 
> how to start it: compile and install kmod-rtl88x2bu execute: `ip link set wlan0 up` and you need to start hostapd manually (if you want AP mode)
> 
> also, netifd ignoring wlan0 in /etc/config/network ifname, so you need to rename interface: for example: add `ip link set wlan0 name wifi0 && ip link set wifi0 up` to /etc/rc.local and add `option ifname 'wifi0'` `option force_link '1'` to your interface (config interface in /etc/config/network) _(maybe not needed for bridges)_

which kernel version

Boos4721, 2022-03-27:
doesn`t work on kernel 5.10-y

Boos4721, 2022-03-27:
on the latest branch, compile error log:
```
SHELL= flock /home/boos/openwrt/tmp/.rtl88x2bu-2022-03-25-1751f39d.tar.xz.flock -c '  	/home/boos/openwrt/scripts/download.pl "/home/boos/openwrt/dl" "rtl88x2bu-2022-03-25-1751f39d.tar.xz" "skip" "" || (  /home/boos/openwrt/scripts/dl_github_archive.py --dl-dir="/home/boos/openwrt/dl" --url="https://github.com/cilynx/rtl88x2bu.git" --version="1751f39dcbc2ab101cb17a09052945613bb9b78c" --subdir="rtl88x2bu-2022-03-25-1751f39d" --source="rtl88x2bu-2022-03-25-1751f39d.tar.xz" --hash="skip" || ( 	echo "Checking out files from the git repository..."; mkdir -p /home/boos/openwrt/tmp/dl && cd /home/boos/openwrt/tmp/dl && rm -rf rtl88x2bu-2022-03-25-1751f39d && [ \! -d rtl88x2bu-2022-03-25-1751f39d ] && git clone  https://github.com/cilynx/rtl88x2bu.git rtl88x2bu-2022-03-25-1751f39d && (cd rtl88x2bu-2022-03-25-1751f39d && git checkout 1751f39dcbc2ab101cb17a09052945613bb9b78c && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd rtl88x2bu-2022-03-25-1751f39d && git log -1 --format='\''@%ct'\''` && rm -rf rtl88x2bu-2022-03-25-1751f39d/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c rtl88x2bu-2022-03-25-1751f39d | 	xz -zc -7e > /home/boos/openwrt/tmp/dl/rtl88x2bu-2022-03-25-1751f39d.tar.xz && mv /home/boos/openwrt/tmp/dl/rtl88x2bu-2022-03-25-1751f39d.tar.xz /home/boos/openwrt/dl/ && rm -rf rtl88x2bu-2022-03-25-1751f39d; );  )    '
rm -f /home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2022-03-25-1751f39d/.built
touch /home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2022-03-25-1751f39d/.built_check
make --jobserver-auth=3,4  -C "/home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.108" KCFLAGS="-fmacro-prefix-map=/home/boos/openwrt/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl" HOSTCFLAGS="-O2 -I/home/boos/openwrt/staging_dir/host/include -I/home/boos/openwrt/staging_dir/hostpkg/include -I/home/boos/openwrt/staging_dir/target-aarch64_generic_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="Boos4721" KBUILD_BUILD_HOST="Boos" KBUILD_BUILD_TIMESTAMP="Sun Mar 27 17:08:23 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/boos/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/boos/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=1  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/antfs.symvers /home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/cryptodev-linux.symvers /home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/gpio-button-hotplug.symvers /home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/mac80211.symvers /home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/mt76.symvers /home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/shortcut-fe.symvers" KERNELRELEASE=5.10.108 M="/home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2022-03-25-1751f39d" NOSTDINC_FLAGS="-I/home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2022-03-25-1751f39d -I/home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2022-03-25-1751f39d/include -I/home/boos/openwrt/staging_dir/target-aarch64_generic_musl/usr/include/mac80211-backport -I/home/boos/openwrt/staging_dir/target-aarch64_generic_musl/usr/include/mac80211-backport/uapi -I/home/boos/openwrt/staging_dir/target-aarch64_generic_musl/usr/include/mac80211 -I/home/boos/openwrt/staging_dir/target-aarch64_generic_musl/usr/include/mac80211/uapi -include backport/autoconf.h -include backport/backport.h" USER_EXTRA_CFLAGS="-DRTW_SINGLE_WIPHY -DRTW_USE_CFG80211_STA_EVENT -DCONFIG_RADIO_WORK -DCONFIG_IOCTL_CFG80211 -DCONFIG_CONCURRENT_MODE -DBUILD_OPENWRT" CONFIG_POWER_SAVING=n CONFIG_RTW_DEBUG=y CONFIG_RTL8822BU=m USER_MODULE_NAME=rtl88x2bu modules
make[3]: Entering directory '/home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.108'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
make -f ./scripts/Makefile.build obj=/home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2022-03-25-1751f39d \
single-build= \
need-builtin=1 need-modorder=1
  aarch64-openwrt-linux-musl-gcc -Wp,-MMD,/home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2022-03-25-1751f39d/core/.rtw_cmd.o.d -I/home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2022-03-25-1751f39d -I/home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2022-03-25-1751f39d/include -I/home/boos/openwrt/staging_dir/target-aarch64_generic_musl/usr/include/mac80211-backport -I/home/boos/openwrt/staging_dir/target-aarch64_gene [... cut ...] 1f39d/core/rtw_mlme.o /home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2022-03-25-1751f39d/core/rtw_mlme.c
make[3]: Leaving directory '/home/boos/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.108'
make[2]: Leaving directory '/home/boos/openwrt/package/kernel/rtl88x2bu'
time: package/kernel/rtl88x2bu/compile#16.71#2.30#4.21
    ERROR: package/kernel/rtl88x2bu failed to build.
make[1]: Leaving directory '/home/boos/openwrt'
```