cilynx/rtl88x2bu #53: Error! Bad return status for module build on kernel: 5.6.0 (aarch64)

gitdy-up, 2020-06-13:
Hi

Installed on my PBP the official Kali from: https://images.offensive-security.com/arm-images/kali-linux-2020.2-pinebook-pro.img.xz

I wanted to install the rtl88x2bu driver for my TP-Link Archer T3U wifi-adapter.

_Ran:_
apt-get update && upgrade

_and then:_ 
git clone https://github.com/cilynx/rtl88x2bu
cd rtl88x2bu
VER=$(sed -n 's/\PACKAGE_VERSION="\(.*\)"/\1/p' dkms.conf)
sudo rsync -rvhP ./ /usr/src/rtl88x2bu-${VER}
sudo dkms add -m rtl88x2bu -v ${VER}
_when running:_
sudo dkms build -m rtl88x2bu -v ${VER}

_I receive the following error:_
[sudo] password for kali: 

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j6 KERNELRELEASE=5.6.0 KVER=5.6.0 src=/usr/src/rtl88x2bu-5.6.1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.6.0 (aarch64)
Consult /var/lib/dkms/rtl88x2bu/5.6.1/build/make.log for more information

_In the /var/lib/dkms/rtl88x2bu/5.6.1/build/make.log I find the following:_
DKMS make.log for rtl88x2bu-5.6.1 for kernel 5.6.0 (aarch64)
Sat Jun 13 19:58:10 UTC 2020
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.6.0/build M=/var/lib/dkms>
make[1]: Entering directory '/usr/src/linux'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are miss>
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[1]: *** [Makefile:686: include/config/auto.conf] Error 1
make[1]: Leaving directory '/usr/src/linux'
make: *** [Makefile:2283: modules] Error 2

I have no ideea how to resolve this. Can you please help?
Thank you!

Medullitus, 2020-06-27:
WHY ANYBODY NOT ANSWERING THESE TOPIC WHERE ARE YOU DEVELOPER WHERE ARE YOU WHERE ???????????????????????????????????????????***

thiagogalvao, 2020-09-19:
Maybe you need to compile for ARM architecture, like a Raspberry PI. 
To compile this drive you need to check your Makefile if the directive CONFIG_PLATFORM_I386_PC is set to false and if CONFIG_PLATFORM_ARM_RPI = y.

This version drive isn't ARM64 compatible.

xMadNet, 2020-10-04:
In the source directory I ran "make oldconfig && make prepare"
But still have after (linking phase)
ERROR: "__const_udelay" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: "__arch_copy_from_user" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: "_raw_spin_unlock" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: "__rcu_read_lock" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: "__rcu_read_unlock" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: "cpu_hwcap_keys" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: "csum_ipv6_magic" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: "__warn_printk" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: "_mcount" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: "arm64_const_caps_ready" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: "__udelay" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: "preempt_schedule_notrace" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: "__arch_copy_to_user" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[1]: *** [Makefile:1645: modules] Error 2
make[1]: Leaving directory '/home/pi/linux-69b14a2e6d4e840c7609370dbf0bac847c3bb15c'
make: *** [Makefile:2283: modules] Error 2

closed 2023-11-02 by MaxG87