cilynx/rtl88x2bu #57: rtl88x2bu: add description of branches to README.md

mhaldrich, 2020-06-17:
Hi, thanks all for maintaining this package.  We use this repository as part of a Yocto OS build to integrate the out of tree module.  

We are moving from the v5.3.1 releases into this `rtl88x2bu` repo.  

Can someone explain to me the difference between these two branches or update the README.md to advise otherse?

cilynx, 2020-06-17:
Howdy -- the branches are for the upstream versions.  I used to create an entirely new repo every time a new version came out upstream, but that got cumbersome with permalink management, so I moved over to a branching strategy.  All of the patches, additions, cleanup, etc in this repo have been committed to the 5.6.1 branch.  Upstream has somewhat recently released a newer 5.6.1.6 version, but as upstream does not follow this repository, all of the changes need to be ported and reconciled with their changes in order to bring 5.6.1.6 to the state of usability where 5.6.1 is.  If you have a better way of stating that in the Administrative Note at the top of the README.md, I'd be happy to accept a PR =)

mhaldrich, 2020-06-23:
@cilynx -- thanks for the response and sorry for the delay.  This makes sense -- you will capture all capture all upstreams and keep them here as branches.  What was unclear is that v5.6.1.6 is still getting patched and updated.  Thanks for clarifying.

I did get a passing build out of the box for the default 5.6.1 branch (against a v5.7.2 kernel in Bitbake), so that was a painless migration!  Ran into some build problems for 5.6.1.6 however and haven't written any updates.

I'll mark this closed, thanks again for maintaining this repo.

closed 2020-06-23 by mhaldrich