cilynx/rtl88x2bu #59: Kernel 5.8 RC2: Error! Bad return status for module build on kernel: 5.8.0-050800-generic (x86_64)

TheJags, 2020-06-26:
Building module fails while trying to install Kernel 5.8 RC2 in Ubuntu MATE 20.04 64-bit with the following error message:

Make and model of the wireless adapter: `TP-Link Archer T3U dual-band USB wireless adapter`

TP Link: `https://www.tp-link.com/us/support/download/archer-t3u/#Driver`

````
Building module:
cleaning build area...
make -j12 KERNELRELEASE=5.8.0-050800-generic KVER=5.8.0-050800-generic src=/usr/src/rtl88x2bu-5.6.1....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.8.0-050800-generic (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.6.1/build/make.log for more information.
   ...done.
````
Here's the **/var/lib/dkms/rtl88x2bu/5.6.1/build/make.log**:

````
DKMS make.log for rtl88x2bu-5.6.1 for kernel 5.8.0-050800-generic (x86_64)

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.8.0-050800-generic/build M=/var/lib/dkms/rtl88x2bu/5.6.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-050800-generic'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_vht.o
In file included from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:79,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_debug.c:17:
/usr/src/rtl88x2bu-5.6.1/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_service_linux.h:48,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_debug.c:17:
./include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:79,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_ioctl_set.c:17:
/usr/src/rtl88x2bu-5.6.1/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_service_linux.h:48,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_ioctl_set.c:17:
./include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:79,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_ieee80211.c:20:
/usr/src/rtl88x2bu-5.6.1/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
[... 165 lines cut ...]
                 from ./include/net/addrconf.h:50,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_service_linux.h:48,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_cmd.c:17:
./include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_query.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ieee80211.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_security.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_set.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_vht.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_io.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_wlan_util.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mi.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_debug.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_cmd.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme_ext.o] Error 1
make[1]: *** [Makefile:1756: /var/lib/dkms/rtl88x2bu/5.6.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-050800-generic'
make: *** [Makefile:2283: modules] Error 2
````

Medullitus, 2020-06-27:
FUCK FUCK FUCK FUCK FUCKKKKKKKKKKKKKKKK

TheJags, 2020-08-02:
Just in case if anyone is interested, I get the same error with Kernel 5.8 RC7 on Ubuntu MATE 20.04 64-bit.

Thanks.

rtl88x2bu_5.6.1_build_make.log:
````
DKMS make.log for rtl88x2bu-5.6.1 for kernel 5.8.0-050800rc7-generic (x86_64)

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.8.0-050800rc7-generic/build M=/var/lib/dkms/rtl88x2bu/5.6.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-050800rc7-generic'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_vht.o

In file included from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:79,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_ieee80211.c:20:
/usr/src/rtl88x2bu-5.6.1/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~

In file included from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:79,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_wlan_util.c:17:
/usr/src/rtl88x2bu-5.6.1/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~

In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_service_linux.h:48,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_ieee80211.c:20:
./include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~

In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_service_linux.h:48,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_wlan_util.c:17:
./include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~

In file included from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:79,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_debug.c:17:
/usr/src/rtl88x2bu-5.6.1/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~

In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
[... 179 lines cut ...]
                 from /usr/src/rtl88x2bu-5.6.1/include/../hal/phydm/phydm_types.h:215,
                 from /usr/src/rtl88x2bu-5.6.1/include/../hal/phydm/phydm_precomp.h:29,
                 from /usr/src/rtl88x2bu-5.6.1/include/hal_data.h:20,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_mlme.c:17:
./include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~

make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_query.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ieee80211.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_security.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_wlan_util.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_set.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_io.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_vht.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mi.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_cmd.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_debug.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme.o] Error 1
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme_ext.o] Error 1
make[1]: *** [Makefile:1756: /var/lib/dkms/rtl88x2bu/5.6.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-050800rc7-generic'
make: *** [Makefile:2283: modules] Error 2
````

HughTebby, 2020-10-26:
Same here with kernel 5.8.0-25-generic

fapinheiro, 2022-01-24:
> Could be a completely different issue here, but I had the same error and it seems I was missing "bc". I installed it with: `sudo apt install bc -y` and it worked after it.

I did what you said and now it´s working. Thank you so much. Tears from Brazil.

Abela-Things, 2024-12-19:
> Could be a completely different issue here, but I had the same error and it seems I was missing "bc". I installed it with: `sudo apt install bc -y` and it worked after it.

Worked for me too, release "5.8.7.1_35809.20191129_COEX20191120-7777" on kernel 5.10.0-33-amd64