cilynx/rtl88x2bu #61: Info: kernel trace on Raspberry PI 3+

hidden-primary-net, 2020-06-28:
Hi,

I just want to let you know of a kernel trace I stumbled upon when using my new "Carantec" USB stick for the very first time:

```
[ 2352.435642] ------------[ cut here ]------------
[ 2352.436254] WARNING: CPU: 0 PID: 911 at /usr/src/rtl88x2bu-5.6.1/hal/hal_com.c:11513 rtw_lps_state_chk+0x64/0x68 [88x2bu]
[ 2352.436262] Modules linked in: 88x2bu(O) rfcomm cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic 8021q garp stp llc evdev joydev brcmfmac brcmutil raspberrypi_hwmon hwmon cfg80211 rfkill bcm2835_codec(C) bcm2835_v4l2(C) snd_bcm2835(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) snd_pcm videobuf2_dma_contig v4l2_common videobuf2_vmalloc snd_timer videobuf2_memops videobuf2_v4l2 snd videobuf2_common videodev media vc_sm_cma(C) rpi_ft5406 uio_pdrv_genirq uio rpi_backlight fixed ip_tables x_tables ipv6
[ 2352.436407] CPU: 0 PID: 911 Comm: RTW_CMD_THREAD Tainted: G         C O      4.19.118-v7+ #1311
[ 2352.436412] Hardware name: BCM2835
[ 2352.436444] [<801120c8>] (unwind_backtrace) from [<8010d5fc>] (show_stack+0x20/0x24)
[ 2352.436461] [<8010d5fc>] (show_stack) from [<80856da8>] (dump_stack+0xe0/0x124)
[ 2352.436479] [<80856da8>] (dump_stack) from [<80120c8c>] (__warn+0x104/0x11c)
[ 2352.436495] [<80120c8c>] (__warn) from [<80120ddc>] (warn_slowpath_null+0x50/0x58)
[ 2352.437043] [<80120ddc>] (warn_slowpath_null) from [<7f884938>] (rtw_lps_state_chk+0x64/0x68 [88x2bu])
[ 2352.438079] [<7f884938>] (rtw_lps_state_chk [88x2bu]) from [<7f8883ec>] (SetHwReg+0x7b4/0x7dc [88x2bu])
[ 2352.439086] [<7f8883ec>] (SetHwReg [88x2bu]) from [<7f8abce4>] (rtl8822b_sethwreg+0xd54/0xda0 [88x2bu])
[ 2352.440096] [<7f8abce4>] (rtl8822b_sethwreg [88x2bu]) from [<7f8af1c0>] (sethwreg+0x1bc/0x1c4 [88x2bu])
[ 2352.441100] [<7f8af1c0>] (sethwreg [88x2bu]) from [<7f87f850>] (rtw_hal_set_hwreg+0x20/0x24 [88x2bu])
[ 2352.442081] [<7f87f850>] (rtw_hal_set_hwreg [88x2bu]) from [<7f838468>] (rtw_set_ps_mode+0xe8/0x228 [88x2bu])
[ 2352.443041] [<7f838468>] (rtw_set_ps_mode [88x2bu]) from [<7f8387c8>] (LPS_Leave+0xd8/0xe0 [88x2bu])
[ 2352.444026] [<7f8387c8>] (LPS_Leave [88x2bu]) from [<7f8a7920>] (rtl8822b_phy_haldm_watchdog+0x1cc/0x1ec [88x2bu])
[ 2352.445032] [<7f8a7920>] (rtl8822b_phy_haldm_watchdog [88x2bu]) from [<7f87feec>] (rtw_hal_dm_watchdog+0x2c/0x30 [88x2bu])
[ 2352.445990] [<7f87feec>] (rtw_hal_dm_watchdog [88x2bu]) from [<7f8030b0>] (rtw_dynamic_chk_wk_hdl+0x40/0x5c [88x2bu])
[ 2352.446910] [<7f8030b0>] (rtw_dynamic_chk_wk_hdl [88x2bu]) from [<7f805174>] (rtw_drvextra_cmd_hdl+0x108/0x3ec [88x2bu])
[ 2352.447826] [<7f805174>] (rtw_drvextra_cmd_hdl [88x2bu]) from [<7f800eb8>] (rtw_cmd_thread+0x188/0x47c [88x2bu])
[ 2352.448293] [<7f800eb8>] (rtw_cmd_thread [88x2bu]) from [<80142b10>] (kthread+0x138/0x168)
[ 2352.448310] [<80142b10>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
[ 2352.448317] Exception stack(0xa8691fb0 to 0xa8691ff8)
[ 2352.448327] 1fa0:                                     00000000 00000000 00000000 00000000
[ 2352.448339] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 2352.448349] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 2352.448436] ---[ end trace 6096a97eec3e1c62 ]---
```

I cloned the repo (branch `5.6.1_30362.20181109_COEX20180928-6a6a`, ) today and cross-compiled the kernel module on a Debian stretch machine (`SMP Debian 4.9.210-1+deb9u1 (2020-06-07) x86_64 GNU/Linux`). If you need more information please let me know.

```
# git show
commit 64aa087c5ead11960425125c955dcfe98233202e
Merge: 0f159d7 3a921c3
Author: Randy C. Will <randall.will@gmail.com>
Date:   Fri Jun 5 06:58:53 2020 -0700

    Merge pull request #51 from ZDima3000/patch-1
    
    fix typo

```

I'm still configuring the machine to become a WiFi client and cannot tell at the moment if it works at all.

hidden-primary-net, 2020-06-28:
The stick works as a WiFi client.

bernhara, 2020-07-13:
Same stack trace on a Fedora/ARM32 installer on a PI2B with a 5.7.9 kernel.
The device is an ULT: https://www.amazon.fr/gp/product/B0823JHCQM.

bernhara, 2020-07-17:
I switched from Fedora to Ubuntu (still arm), and now the driver is absolutely reliable.
My guess is that the way the USB power is driven by the distro has an impact: when I configured my Wifi device to AP (=> power increase), I had more problems.
Instead of investigating an eventual power problem, I changed my Linux distribution.

anajilly, 2020-09-04:
I get a lot of stack traces very much like that one, on ubuntu 18.04 with `uname -r` == 4.15.0-112-generic

my system has crashed once since building/installing this driver, and the logged stack traces from this driver seem to be the biggest clue (though I have probably had problems with my video card too).

typical stack trace:
```
Sep 03 22:45:52 laverne kernel: ------------[ cut here ]------------
Sep 03 22:45:52 laverne kernel: WARNING: CPU: 6 PID: 1161 at /usr/src/rtl88x2bu-5.6.1/hal/hal_com.c:11513 rtw_lps_state_chk+0x45/0x4e [88x2bu]
Sep 03 22:45:52 laverne kernel: Modules linked in: vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) snd_hda_codec_hdmi binfmt_misc intel_rapl skx_edac hp_wmi nvidia_drm(POE) x86_pkg_temp_thermal intel_powerclamp nvidia_modeset(POE) nls_iso8859_1 sparse_keymap intel_wmi_thunderbolt wmi_bmof coretemp kvm_intel snd_hda_codec_realtek snd_hda_codec_generic 88x2bu(OE) kvm irqbypass crct10dif_pclmul snd_seq_midi crc32_pclmul nvidia(POE) ghash_clmulni_intel snd_seq_midi_event pcbc snd_rawmidi snd_hda_intel aesni_intel snd_hda_codec aes_x86_64 crypto_simd glue_helper cryptd cdc_ether snd_hda_core usbnet snd_seq intel_cstate r8152 snd_hwdep cfg80211 mii intel_rapl_perf joydev input_leds drm_kms_helper snd_seq_device serio_raw snd_pcm drm ipmi_devintf snd_timer ipmi_msghandler fb_sys_fops syscopyarea sysfillrect sysimgblt snd mei_me
Sep 03 22:45:52 laverne kernel:  soundcore mei ioatdma shpchp wmi mac_hid sch_fq_codel parport_pc ppdev lp parport nfsd auth_rpcgss nfs_acl lockd grace sunrpc ip_tables x_tables autofs4 hid_generic usbhid hid nvme psmouse ixgbe igb nvme_core e1000e i2c_algo_bit mdio dca ptp ahci pps_core libahci
Sep 03 22:45:52 laverne kernel: CPU: 6 PID: 1161 Comm: RTW_CMD_THREAD Tainted: P        W  OE    4.15.0-112-generic #113-Ubuntu
Sep 03 22:45:52 laverne kernel: Hardware name: HP HP Z4 G4 Workstation/81C5, BIOS P61 v02.48 03/27/2020
Sep 03 22:45:52 laverne kernel: RIP: 0010:rtw_lps_state_chk+0x45/0x4e [88x2bu]
Sep 03 22:45:52 laverne kernel: RSP: 0018:ffffba1e02ea3c90 EFLAGS: 00010286
Sep 03 22:45:52 laverne kernel: RAX: 0000000000000024 RBX: 0000000000000000 RCX: 0000000000000006
Sep 03 22:45:52 laverne kernel: RDX: 0000000000000000 RSI: ffff9b3dbfb96498 RDI: ffff9b3dbfb96490
Sep 03 22:45:52 laverne kernel: RBP: ffffba1e02ea3ca0 R08: 0000000000000656 R09: 0000000000000004
Sep 03 22:45:52 laverne kernel: R10: 0000000000000000 R11: 0000000000000001 R12: ffffba1e02139000
Sep 03 22:45:52 laverne kernel: R13: ffffba1e02ea3d6c R14: ffffba1e0285d000 R15: ffff9b3da5269be0
Sep 03 22:45:52 laverne kernel: FS:  0000000000000000(0000) GS:ffff9b3dbfb80000(0000) knlGS:0000000000000000
Sep 03 22:45:52 laverne kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Sep 03 22:45:52 laverne kernel: CR2: 00007f4fd95f5514 CR3: 000000015c60a002 CR4: 00000000003606e0
Sep 03 22:45:52 laverne kernel: DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
Sep 03 22:45:52 laverne kernel: DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
Sep 03 22:45:52 laverne kernel: Call Trace:
Sep 03 22:45:52 laverne kernel:  SetHwReg+0x704/0x726 [88x2bu]
Sep 03 22:45:52 laverne kernel:  rtl8822b_sethwreg+0xdec/0xe02 [88x2bu]
Sep 03 22:45:52 laverne kernel:  sethwreg+0x134/0x13b [88x2bu]
Sep 03 22:45:52 laverne kernel:  rtw_hal_set_hwreg+0x19/0x1b [88x2bu]
Sep 03 22:45:52 laverne kernel:  rtw_set_ps_mode+0xf3/0x223 [88x2bu]
Sep 03 22:45:52 laverne kernel:  LPS_Leave+0xd0/0xd7 [88x2bu]
Sep 03 22:45:52 laverne kernel:  rtl8822b_phy_haldm_watchdog+0x16b/0x191 [88x2bu]
Sep 03 22:45:52 laverne kernel:  rtw_hal_dm_watchdog+0x21/0x24 [88x2bu]
Sep 03 22:45:52 laverne kernel:  rtw_dynamic_chk_wk_hdl+0x37/0x52 [88x2bu]
Sep 03 22:45:52 laverne kernel:  rtw_drvextra_cmd_hdl+0x2a0/0x496 [88x2bu]
Sep 03 22:45:52 laverne kernel:  ? _rtw_up_sema+0xe/0x10 [88x2bu]
Sep 03 22:45:52 laverne kernel:  rtw_cmd_thread+0x1ba/0x454 [88x2bu]
Sep 03 22:45:52 laverne kernel:  ? session_tracker_cmd_hdl+0x1df/0x1df [88x2bu]
Sep 03 22:45:52 laverne kernel:  kthread+0x121/0x140
Sep 03 22:45:52 laverne kernel:  ? rtw_stop_cmd_thread+0x3f/0x3f [88x2bu]
Sep 03 22:45:52 laverne kernel:  ? kthread_create_worker_on_cpu+0x70/0x70
Sep 03 22:45:52 laverne kernel:  ret_from_fork+0x35/0x40
Sep 03 22:45:52 laverne kernel: Code: 04 06 00 00 4c 89 e7 e8 35 24 f9 ff 84 c0 79 1d bf 01 00 00 00 e8 83 69 fd ff 80 eb 01 75 e0 48 c7 c7 78 6b 13 c2 e8 23 c8 67 db <0f> 0b 5b 41 5c 5d c3 f3 c3 0f 1f 44 00 00 55 48 89 e5 41 54 53
Sep 03 22:45:52 laverne kernel: ---[ end trace fb25e3a31bfab7ff ]---
```

is there anything I can do to help? Is there any documentation available for the hardware?

bggardner, 2020-09-20:
I'm getting these stack trace dumps quite frequently on a Raspberry Pi 2 Model B Rev 1.1.  My adapter is no longer listed on Amazon, but similar to this one: https://www.amazon.com/dp/B07W47TVM6/

I'm wondering if @bernhara is right, so I'll try to find a powered USB hub to see if it's power-related.