cilynx/rtl88x2bu #63: Access point with 2.4 and 5 GHz

sphh, 2020-07-06:
I have a TP-Link Archer AC1300 USB dongle and use it for an access point with 5 GHz on my RPi 1. Works great, thanks!

I wondered, if it were possible to make an access point on 2.4 _and_ 5 GHz at the same time?

AleXSR700, 2020-08-07:
I will ammend the question and ask: and maybe support 867 MBit/s in AP mode? :)

deepsidhu1313, 2020-08-29:
@cilynx is there a way to enable 867mbps in ap mode using these drivers ?

deepsidhu1313, 2020-08-30:
I was able to create 300mbps link using the following config
```
ssid=your_ssid
wpa_passphrase=your_password

country_code=US

interface=wlan1
driver=nl80211

wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP

macaddr_acl=0

logger_syslog=0
logger_syslog_level=4
logger_stdout=-1
logger_stdout_level=0

hw_mode=a
wmm_enabled=1

# N
ieee80211n=1
require_ht=1
ht_capab=[MAX-AMSDU-3839][HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]

# AC
ieee80211ac=1
require_vht=1
ieee80211d=0
ieee80211h=0
vht_capab=[MAX-AMSDU-3839][SHORT-GI-80]
vht_oper_chwidth=1
channel=36
vht_oper_centr_freq_seg0_idx=42
```

```
$ iwconfig
wlan1     IEEE 802.11bgn  ESSID:"your_ssid"  Nickname:"<WIFI@REALTEK>"
          Mode:Master  Frequency:5.18 GHz  Access Point: XX:XX:XX:XX:XX:XX   
          Bit Rate:300 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=1/100  Signal level=-99 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

```
$ iw list
Wiphy phy1
    max # scan SSIDs: 9
    max scan IEs length: 2304 bytes
    max # sched scan SSIDs: 0
    max # match sets: 0
    max # scan plans: 1
    max scan plan interval: -1
    max scan plan iterations: 0
    Retry short limit: 7
    Retry long limit: 4
    Coverage class: 0 (up to 0m)
    Supported Ciphers:
        * WEP40 (00-0f-ac:1)
        * WEP104 (00-0f-ac:5)
        * TKIP (00-0f-ac:2)
        * CCMP-128 (00-0f-ac:4)
[... 146 lines cut ...]
         * connect
         * disconnect
    Supported TX frame types:
         * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
         * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
         * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
         * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
         * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
         * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
    Supported RX frame types:
         * IBSS: 0xd0
         * managed: 0x40 0xb0 0xd0
         * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
         * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
         * P2P-client: 0x40 0xd0
         * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
    WoWLAN support:
         * wake up on anything (device continues operating normally)
    software interface modes (can always be added):
         * monitor
    interface combinations are not supported
    Device supports SAE with AUTHENTICATE command
    Device supports scan flush.
    Supported extended features:
```

sphh, 2020-08-31:
@deepsidhu1313 thanks for sharing.

Can you connect both with 2.4 and 5 GHz at the **_same_** time?

deepsidhu1313, 2020-09-13:
@sphh  no apparently it doesn't work that way. As per [Gentoo Wiki](https://wiki.gentoo.org/wiki/Hostapd)

> An AP can only use one band at a time: 2.4GHz OR 5GHz, a so-called "dual-band AP" is just one AP at 2.4GHz and another at 5GHz;

sphh, 2020-09-14:
Thanks @deepsidhu1313! But it also [says](https://wiki.gentoo.org/wiki/Hostapd#What_it_can_do):

> Use 2.4GHz and 5GHz at the same time on the same card.

with the caveat:

> … This requires a card with two radios though, which is pretty rare (but hostapd supports it) - if the card creates two wlanX interfaces, you might be lucky;

I guess I am just not lucky with my card! It creates only one `wlanX` interface, so I have to assume, that my card is not one of that rare breed.

@AleXSR700, I will close this issue, because my original question was answered. Maybe you want to open another issue with your question?

Just for the record: I already resorted to use a second card and call `hostapd` with two configuration files. On the Raspbian Buster OS, do the following:
1. Get two cards (USB WiFi Sticks), one for 2.4GHz and one for 5GHz.
2. Write two separate hostapd configuration files for both APs and put them into `/etc/hostapd/`. I called them `/etc/hostapd/ap_2_4GHz.conf` and `/etc/hostapd/ap_5GHz.conf`.
3. Edit `/etc/default/hostapd` and change `DAEMON_CONF` to contain both files, e.g. `DAEMON_CONF="/etc/hostapd/ap_2_4GHz.conf /etc/hostapd/ap_5GHz.conf"`.
4. Work around [this bug](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=919892):
`sudo systemctl edit hostapd.service`
and add the following contents:
```
[Service]
ExecStart=
ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid $DAEMON_OPTS $DAEMON_CONF
```
(The difference to the original service file is the missing curly brackets in `${DAEMON_CONF}`!

5. Restart hostapd
`sudo systemctl restart hostapd.service`

closed 2020-09-14 by sphh