cilynx/rtl88x2bu #64: Wifi network hangs periodically, requires restart of network (Ubuntu 20.04)

adefaria, 2020-07-09:
For a few months now I've experienced outages where all of a sudden all network activity fails. This happens only on my desktop machine that uses this driver. My other systems in the house can get out to the net with no issues and they are all talking to the same wireless router. The solution I stumbled into is simply toggling the wifi off then back on in NetworkManager and I'm back. But the occurrences are too many and bothersome. I had hoped with a new kernel and perhaps a new driver the problem would go away but it got worse.

rgpublic, 2021-02-20:
For me the solution was to edit:

`/etc/NetworkManager/conf.d/default-wifi-powersave-on.conf`

and set:

```
[connection]
wifi.powersave = 2
```
This made the driver very stable for me. Without that change, I also had connection problems outages etc. I think these problems only happen in conjunction with NetworkManager.

adefaria, 2021-03-02:
My solution was to get a new desktop. I was due. I now have a new System76 Theilo and it's great. No problems.

closed 2021-03-02 by adefaria