cilynx/rtl88x2bu #70: Ubuntu 20.04 kernel 5.4.0.42-generic not responding on modprobe.

ktmlleska, 2020-07-29:
Hi,

I'm trying to install the drivers for my Asus nano AC-53 on the ubuntu 20.04 but when I try to sudo modprobe 8822bu nothing happens, is like the terminal gets freeze.

Do you know any solution on this?

thanks.

githolgi, 2020-08-14:
I initially ran into the same problem (having similar setup, but running in a vm).
I left out the last line of installation script containing the modprobe, and rebooted instead.
The driver works after reboot.

ktmlleska, 2020-08-17:
It's all working now, it was just a matter of re-install ubuntu with a cable and internet connection. The problem was an error with the PCI card, that has been fixed on the re install process.

Thanks.

closed 2020-08-17 by ktmlleska