cilynx/rtl88x2bu #72: Build failed in Linux kernel 5.8

SuzukiHonoka, 2020-08-06:
logs
`
In file included from /mnt/ramdisk/rtl88x2bu/include/drv_types.h:79:0,
                 from /mnt/ramdisk/rtl88x2bu/core/rtw_io.c:47:
/mnt/ramdisk/rtl88x2bu/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
 struct sha256_state {
        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23:0,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /mnt/ramdisk/rtl88x2bu/include/osdep_service_linux.h:48,
                 from /mnt/ramdisk/rtl88x2bu/include/osdep_service.h:50,
                 from /mnt/ramdisk/rtl88x2bu/include/drv_types.h:27,
                 from /mnt/ramdisk/rtl88x2bu/core/rtw_io.c:47:
./include/crypto/sha.h:84:8: note: originally defined here
 struct sha256_state {
        ^~~~~~~~~~~~
In file included from /mnt/ramdisk/rtl88x2bu/include/drv_types.h:79:0,
                 from /mnt/ramdisk/rtl88x2bu/core/rtw_security.c:17:
/mnt/ramdisk/rtl88x2bu/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
 struct sha256_state {
        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23:0,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /mnt/ramdisk/rtl88x2bu/include/osdep_service_linux.h:48,
                 from /mnt/ramdisk/rtl88x2bu/include/osdep_service.h:50,
                 from /mnt/ramdisk/rtl88x2bu/include/drv_types.h:27,
                 from /mnt/ramdisk/rtl88x2bu/core/rtw_security.c:17:
./include/crypto/sha.h:84:8: note: originally defined here
 struct sha256_state {
        ^~~~~~~~~~~~
In file included from /mnt/ramdisk/rtl88x2bu/include/drv_types.h:79:0,
                 from /mnt/ramdisk/rtl88x2bu/core/rtw_ieee80211.c:20:
/mnt/ramdisk/rtl88x2bu/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
 struct sha256_state {
        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23:0,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /mnt/ramdisk/rtl88x2bu/include/osdep_service_linux.h:48,
                 from /mnt/ramdisk/rtl88x2bu/include/osdep_service.h:50,
                 from /mnt/ramdisk/rtl88x2bu/include/drv_types.h:27,
                 from /mnt/ramdisk/rtl88x2bu/core/rtw_ieee80211.c:20:
./include/crypto/sha.h:84:8: note: originally defined here
 struct sha256_state {
        ^~~~~~~~~~~~
In file included from /mnt/ramdisk/rtl88x2bu/include/drv_types.h:79:0,
                 from /mnt/ramdisk/rtl88x2bu/core/rtw_debug.c:17:
/mnt/ramdisk/rtl88x2bu/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
 struct sha256_state {
        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23:0,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /mnt/ramdisk/rtl88x2bu/include/osdep_service_linux.h:48,
                 from /mnt/ramdisk/rtl88x2bu/include/osdep_service.h:50,
                 from /mnt/ramdisk/rtl88x2bu/include/drv_types.h:27,
                 from /mnt/ramdisk/rtl88x2bu/core/rtw_debug.c:17:
./include/crypto/sha.h:84:8: note: originally defined here
 struct sha256_state {
        ^~~~~~~~~~~~
[... 187 lines cut ...]
                 from /mnt/ramdisk/rtl88x2bu/include/osdep_service.h:50,
                 from /mnt/ramdisk/rtl88x2bu/include/drv_types.h:27,
                 from /mnt/ramdisk/rtl88x2bu/core/rtw_recv.c:17:
./include/crypto/sha.h:84:8: note: originally defined here
 struct sha256_state {
        ^~~~~~~~~~~~
In file included from /mnt/ramdisk/rtl88x2bu/include/drv_types.h:79:0,
                 from /mnt/ramdisk/rtl88x2bu/core/rtw_chplan.c:17:
/mnt/ramdisk/rtl88x2bu/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
 struct sha256_state {
        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23:0,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /mnt/ramdisk/rtl88x2bu/include/osdep_service_linux.h:48,
                 from /mnt/ramdisk/rtl88x2bu/include/osdep_service.h:50,
                 from /mnt/ramdisk/rtl88x2bu/include/drv_types.h:27,
                 from /mnt/ramdisk/rtl88x2bu/core/rtw_chplan.c:17:
./include/crypto/sha.h:84:8: note: originally defined here
 struct sha256_state {
        ^~~~~~~~~~~~
`
plz take a look, thx~

SuzukiHonoka, 2020-08-06:
#58 This patch works for me.

TheJags, 2020-08-06:
@SuzukiHonoka 

Could you please tell me how to apply this patch / pull request #58 `https://github.com/cilynx/rtl88x2bu/pull/58` to the master branch (that I have downloaded already through "git clone"). I'm using Ubuntu MATE 20.04

By the way, I've reported the same issue too: `https://github.com/cilynx/rtl88x2bu/issues/59`

SuzukiHonoka, 2020-08-07:
@TheJags 
Hi,
You can simply add a surrfix to the pull request URL to get the entire patch file.
```
wget https://github.com/cilynx/rtl88x2bu/pull/58.patch
git apply 58.patch
```

TheJags, 2020-08-07:
@SuzukiHonoka

Thanks alot for the reply. Is this how it should be:

````
git clone https://github.com/cilynx/rtl88x2bu.git
cd rtl88x2bu
wget https://github.com/cilynx/rtl88x2bu/pull/58.patch
git apply 58.patch
````
I just like to make sure as there was no message after, `git apply 58.patch` (the prompt just went to next line).

SuzukiHonoka, 2020-08-07:
Yep, that's right~

TheJags, 2020-08-07:
> Yep, that's right~
Thank you :)

StaticallyTypedRice, 2020-09-01:
I'm having an issue with kernel `5.8.4-200-fc32-x86_64` on Fedora 32 where the DKMS module (with patch 58 applied) installs but the Wi-Fi adapter is still not detected.

`sudo dkms status` outputs the following: `rtl88x2bu, 5.6.1, 5.8.4-200-fc32.x86_64, x86_64: installed`

closed 2020-12-16 by SuzukiHonoka