cilynx/rtl88x2bu #75: build failed in 5.7.0-kali1-amd64

MiChaelinzo, 2020-08-30:
[image]
[image]

gabe-swp, 2020-09-01:
I'm also getting this exact error. Do you know an earlier version of Kali I could use whilst I wait for this to be fixed?

TheThunderingMarmot, 2020-09-06:
This problem also happens on kernel 5.5, it is not a driver problem.
The fix is installing a package called "bc" or at least, that's what worked for me.

MiChaelinzo, 2020-09-07:
> This problem also happens on kernel 5.5, it is not a driver problem.
> The fix is installing a package called "bc" or at least, that's what worked for me.

sudo apt-get install bc?

TheThunderingMarmot, 2020-09-07:
> > This problem also happens on kernel 5.5, it is not a driver problem.
> > The fix is installing a package called "bc" or at least, that's what worked for me.
> 
> sudo apt-get install bc?
> https://www.gnu.org/software/bc/

Yep, that's it.

MiChaelinzo, 2020-09-11:
I still can't compile driver source with 
$ make clean
$ make
I have Installed the Kernel Header File
automatically with $ sudo apt-get install linux-headers-$(uname -r)
I don't know if the instruction for Linux in TP-Link website is outdated or it doesn't have instruction for Debian

closed 2021-01-29 by MaxG87