cilynx/rtl88x2bu #78: Info: possibly newer driver available from trendtechcn.com

KnallbertLp, 2020-09-07:
I found a apparently new driver ( rtl88x2bu-5.8.7.2.36899.20200819 ) at [deb.trendtechcn.com](https://deb.trendtechcn.com)
I followed their [manual installation](https://deb.trendtechcn.com/advanced/manual-installation/) and found this newer version of the driver

`kb@UnknownPlane:/tmp/tmp.AJ9b0pRFd2$ tree
.
├── control.tar.gz
├── data.tar.gz
├── debian-binary
├── rtl88x2bu-dkms.deb
└── usr
    ├── lib
    │   └── NetworkManager
    │       └── conf.d
    │           └── rtl88x2bu.conf
    ├── share
    │   ├── doc
    │   │   └── rtl88x2bu-dkms
    │   │       ├── changelog.Debian.gz
    │   │       └── copyright
    │   └── rtl88x2bu-dkms
    │       ├── trendtechcn.gpg
    │       └── trendtechcn.list
    └── src
        └── rtl88x2bu-5.8.7.2.36899.20200819
            ├── Kconfig
            ├── Makefile
            ├── core
            │   ├── efuse
            │   │   └── rtw_efuse.c
            │   ├── mesh
            │   │   ├── rtw_mesh.c
            │   │   ├── rtw_mesh.h
            │   │   ├── rtw_mesh_hwmp.c
            │   │   ├── rtw_mesh_hwmp.h
            │   │   ├── rtw_mesh_pathtbl.c
            │   │   └── rtw_mesh_pathtbl.h
            │   ├── rtw_ap.c
            │   ├── rtw_beamforming.c
            │   ├── rtw_br_ext.c
            │   ├── rtw_bt_mp.c
            │   ├── rtw_btcoex.c
            │   ├── rtw_btcoex_wifionly.c
            │   ├── rtw_chplan.c
            │   ├── rtw_chplan.h
            │   ├── rtw_cmd.c
            │   ├── rtw_debug.c
            │   ├── rtw_eeprom.c
            │   ├── rtw_ieee80211.c
            │   ├── rtw_io.c
            │   ├── rtw_ioctl_query.c
            │   ├── rtw_ioctl_set.c
            │   ├── rtw_iol.c
            │   ├── rtw_mem.c
            │   ├── rtw_mi.c
            │   ├── rtw_mlme.c
            │   ├── rtw_mlme_ext.c
            │   ├── rtw_mp.c
            │   ├── rtw_odm.c
            │   ├── rtw_p2p.c
            │   ├── rtw_pwrctrl.c
            │   ├── rtw_recv.c
            │   ├── rtw_rf.c
            │   ├── rtw_rm.c
            │   ├── rtw_rm_fsm.c
            │   ├── rtw_rm_util.c
            │   ├── rtw_rson.c
            │   ├── rtw_sdio.c
            │   ├── rtw_security.c
            │   ├── rtw_sreset.c
            │   ├── rtw_sta_mgt.c
[... 620 lines cut ...]
            │   ├── platform_sprd_sdio.c
            │   ├── platform_zte_zx296716_sdio.c
            │   └── platform_zte_zx296716_sdio.h
            └── rtl8822b.mk

36 directories, 653 files`

I thougth this might be a helpful addition to this communitiy created driver. 

I followed part of their [manual installation](https://deb.trendtechcn.com/advanced/manual-installation/) instructions with the .deb file in [rtl88x2bu-dkms.zip](https://github.com/cilynx/rtl88x2bu/files/5182449/rtl88x2bu-dkms.zip)

```
Getting the source and compiling

From our driver page, retrieve the .deb package that matches your adapter and uncompress it to get the source. Here’s an example for AC1L V2 and AC3L V2:

# Create a temporary folder
cd $(mktemp -d)
# Get the .deb
wget https://deb.trendtechcn.com/rtl88x2bu-dkms.deb

# Uncompress the .deb
ar x rtl88x2bu-dkms.deb
tar xf data.tar.gz .
```

krash1220, 2020-09-09:
Thank you for this. I installed it and I think it made it even faster.

KnallbertLp, 2020-09-09:
If somebody could review this driver and check it for security risks / backdoors that would be beautiful.
I am kinda suspicious because the provider is a Chinese company

vladocou, 2020-09-10:
Tried it in Debian bullseye (kernel: 5.7.0-3-amd64) for a TP-LINK T3U Plus (US) Ver: 1.0 adapter.
Compilation worked with this warning: 
`  CC [M]  /home/MYUSER/Downloads/trendtech_driver/usr/src/rtl88x2bu-5.8.7.2.36899.20200819/os_dep/linux/rtw_proc.o
/home/MYUSER/Downloads/trendtech_driver/usr/src/rtl88x2bu-5.8.7.2.36899.20200819/os_dep/linux/ioctl_cfg80211.c:4311:20: warning: ‘nl80211_tx_power_setting_str’ defined but not used [-Wunused-function]
 4311 | static const char *nl80211_tx_power_setting_str(int type)
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~

  CC [M]  /home/MYUSER/Downloads/trendtech_driver/usr/src/rtl88x2bu-5.8.7.2.36899.20200819/os_dep/linux/rtw_rhashtable.o
`
But after loading it with insmod nothing happens. I mean, the module IS loaded but the adapter won't use it.

After that I tried installing the .deb (with dkms support). Installation went OK but again, device won't even turn on the LED. Checking journalctl I noticed warnings about "Undefined symbols". If they're useful I can provide them.

cilynx, 2020-09-10:
@vladocou -- The T3U Plus isn't included in the ID list by default.  Try adding it in the new driver.  See 5a0c5c872ce505cef9fab938c6ada3c159e2ed9a for details.

vladocou, 2020-09-10:
> @vladocou -- The T3U Plus isn't included in the ID list by default. Try adding it in the new driver. See [5a0c5c8](https://github.com/cilynx/rtl88x2bu/commit/5a0c5c872ce505cef9fab938c6ada3c159e2ed9a) for details.

Thanks for the advice! I did that and I got the adapter up and scanning, but it couldn't establish a connection. Looking up journalctl I got this: 
<details>
  <summary>Click to expand a journalctl wall of text!</summary>

```bash
sep 10 19:32:28 zonda kernel: usb 1-4: new high-speed USB device number 12 using xhci_hcd
sep 10 19:32:28 zonda kernel: usb 1-4: New USB device found, idVendor=2357, idProduct=0138, bcdDevice= 2.10
sep 10 19:32:28 zonda kernel: usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
sep 10 19:32:28 zonda kernel: usb 1-4: Product: 802.11ac NIC
sep 10 19:32:28 zonda kernel: usb 1-4: Manufacturer: Realtek
sep 10 19:32:28 zonda kernel: usb 1-4: SerialNumber: 123456
sep 10 19:32:43 zonda NetworkManager[646]: <info>  [1599777163.6922] device (wlan0): driver supports Access Point (AP) mode
sep 10 19:32:43 zonda NetworkManager[646]: <info>  [1599777163.6929] manager: (wlan0): new 802.11 Wi-Fi device (/org/freedesktop/NetworkManager/Devices/10)
sep 10 19:32:43 zonda systemd[1]: Starting Load/Save RF Kill Switch Status...
sep 10 19:32:43 zonda mtp-probe[12956]: checking bus 1, device 12: "/sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-4"
sep 10 19:32:43 zonda mtp-probe[12956]: bus: 1, device: 12 was not an MTP device
sep 10 19:32:43 zonda systemd-udevd[12958]: Using default interface naming scheme 'v245'.
sep 10 19:32:43 zonda NetworkManager[646]: <info>  [1599777163.6978] rfkill4: found Wi-Fi radio killswitch (at /sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-4/1-4:1.0/ieee80211/phy4/rfkill4) (driver rtl88x2bu)
sep 10 19:32:43 zonda systemd-udevd[12958]: ethtool: autonegotiation is unset or enabled, the speed and duplex are not writable.
sep 10 19:32:43 zonda kernel: rtl88x2bu 1-4:1.0 wlx9848273b2c09: renamed from wlan0
sep 10 19:32:43 zonda systemd[1]: Started Load/Save RF Kill Switch Status.
sep 10 19:32:43 zonda NetworkManager[646]: <info>  [1599777163.7221] device (wlan0): interface index 9 renamed iface from 'wlan0' to 'wlx9848273b2c09'
sep 10 19:32:43 zonda NetworkManager[646]: <info>  [1599777163.7314] device (wlx9848273b2c09): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
sep 10 19:32:43 zonda mtp-probe[12974]: checking bus 1, device 12: "/sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-4"
sep 10 19:32:43 zonda mtp-probe[12974]: bus: 1, device: 12 was not an MTP device
sep 10 19:32:43 zonda systemd-udevd[12969]: Using default interface naming scheme 'v245'.
sep 10 19:32:46 zonda ModemManager[681]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-4': not supported by any plugin
sep 10 19:32:48 zonda systemd[1]: systemd-rfkill.service: Succeeded.
sep 10 19:32:56 zonda kernel: start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
sep 10 19:32:57 zonda wpa_supplicant[12949]: nl80211: kernel reports: Authentication algorithm number required
sep 10 19:32:57 zonda NetworkManager[646]: <info>  [1599777177.7780] device (wlx9848273b2c09): supplicant interface state: internal-starting -> disconnected
sep 10 19:32:57 zonda NetworkManager[646]: <info>  [1599777177.7781] Wi-Fi P2P device controlled by interface wlx9848273b2c09 created
sep 10 19:32:57 zonda NetworkManager[646]: <info>  [1599777177.7783] manager: (p2p-dev-wlx9848273b2c09): new 802.11 Wi-Fi P2P device (/org/freedesktop/NetworkManager/Devices/11)
sep 10 19:32:57 zonda NetworkManager[646]: <info>  [1599777177.7786] device (p2p-dev-wlx9848273b2c09): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
sep 10 19:32:57 zonda NetworkManager[646]: <info>  [1599777177.7792] device (wlx9848273b2c09): state change: unavailable -> disconnected (reason 'supplicant-available', sys-iface-state: 'managed')
sep 10 19:32:57 zonda NetworkManager[646]: <info>  [1599777177.7798] device (p2p-dev-wlx9848273b2c09): state change: unavailable -> disconnected (reason 'none', sys-iface-state: 'managed')
sep 10 19:32:57 zonda NetworkManager[646]: <warn>  [1599777177.7839] sup-iface[07b343f5aa1b0b2a,0,wlx9848273b2c09]: call-p2p-cancel: failed with P2P cancel failed
sep 10 19:33:00 zonda ModemManager[681]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-4': not supported by any plugin
sep 10 19:33:10 zonda NetworkManager[646]: <info>  [1599777190.9888] policy: auto-activating connection 'A Nisman' (1d9b1f39-0251-4e74-ba51-9b54492410f1)
sep 10 19:33:10 zonda NetworkManager[646]: <info>  [1599777190.9892] device (wlx9848273b2c09): Activation: starting connection 'A Nisman' (1d9b1f39-0251-4e74-ba51-9b54492410f1)
sep 10 19:33:10 zonda NetworkManager[646]: <info>  [1599777190.9893] device (wlx9848273b2c09): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
sep 10 19:33:10 zonda NetworkManager[646]: <info>  [1599777190.9965] device (wlx9848273b2c09): set-hw-addr: set-cloned MAC address to 8A:E7:F5:5C:42:81 (8A:E7:F5:5C:42:81)
sep 10 19:33:10 zonda NetworkManager[646]: <info>  [1599777190.9968] device (wlx9848273b2c09): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
[... cut ...]
sep 10 19:34:06 zonda wpa_supplicant[12949]: nl80211: kernel reports: Authentication algorithm number required
```
</details> 

I think these two lines could be relevant and I should mention that I got the same error while trying another module: 

```bash
sep 10 19:33:42 zonda wpa_supplicant[12949]: wlx9848273b2c09: Trying to associate with 64:70:02:a1:ee:7c (SSID='A Nisman' freq=2447 MHz)
sep 10 19:33:42 zonda wpa_supplicant[12949]: nl80211: kernel reports: Authentication algorithm number required
```

krash1220, 2020-09-11:
I've had this problem and all I did was reboot the modem and everything was good after that but your problem may be different.

vladocou, 2020-09-11:
> I've had this problem and all I did was reboot the modem and everything was good after that but your problem may be different.

Tried that, didn't work. Also I tried to connect to my phone in hotspot mode without the need for authentication and that didn't work either. 

So yeah, tomorrow I'll try a few more drivers and if I don't have any luck I guess I'll have to return the adapter :/

krash1220, 2020-09-11:
If you want an excellent adapter, this is the one I bought. It's very fast. I'm getting about 260 Mbps which is about 130 Mbps more than I was getting with my on-board adapter.  I downloaded the deb version of this driver and all I had to do was click on it and install it. It shows up in Drive Manager in Linux Mint. Don't know what distro you are working on or what your machine is but I'm very happy with this USB adapter.

https://www.amazon.com/gp/product/B07FCN6WGX/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1

vladocou, 2020-09-11:
> If you want an excellent adapter, this is the one I bought. It's very fast. I'm getting about 260 Mbps which is about 130 Mbps more than I was getting with my on-board adapter. I downloaded the deb [...]
I'm running Debian bullseye (kernel: 5.7.0-3-amd64).
Thank you very much for your suggestion! Sadly, importing things like that to Argentina is a) Prohibitively expensive (taxes) or b) The seller just don't ship it (very often the packages are stolen in customs).

closed 2021-07-18 by KnallbertLp