cilynx/rtl88x2bu #85: Error! Bad return status for module build on kernel on Kali linux

xdelu02, 2020-09-30:
sudo dkms build -m rtl88x2bu -v ${VER}

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.8.0-kali2-amd64 KVER=5.8.0-kali2-amd64 src=/usr/src/rtl88x2bu-5.6.1....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.8.0-kali2-amd64 (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.6.1/build/make.log for more information.

0xCAP, 2020-10-12:
Getting almost the same error on 5.8.14-arch1-1

gioulisapo, 2020-10-15:
Hello, 
I get the same issue, cant use the dongle on kali unfortunately as a result. I attach the logs if they are of any help:
```
> cat /var/lib/dkms/rtl88x2bu/5.6.1/build/make.log
DKMS make.log for rtl88x2bu-5.6.1 for kernel 5.8.0-kali2-amd64 (x86_64)
Thu 15 Oct 19:18:53 BST 2020
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.8.0-kali2-amd64/build M=/var/lib/dkms/rtl88x2bu/5.6.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-kali2-amd64'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme.o
In file included from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:79,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_security.c:17:
/usr/src/rtl88x2bu-5.6.1/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from /usr/src/linux-headers-5.8.0-kali2-common/include/linux/filter.h:23,
                 from /usr/src/linux-headers-5.8.0-kali2-common/include/net/sock.h:59,
                 from /usr/src/linux-headers-5.8.0-kali2-common/include/linux/tcp.h:19,
                 from /usr/src/linux-headers-5.8.0-kali2-common/include/linux/ipv6.h:88,
                 from /usr/src/linux-headers-5.8.0-kali2-common/include/net/addrconf.h:50,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_service_linux.h:48,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_security.c:17:
/usr/src/linux-headers-5.8.0-kali2-common/include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:79,
                 from /usr/src/rtl88x2bu-5.6.1/include/../hal/phydm/phydm_types.h:215,
                 from /usr/src/rtl88x2bu-5.6.1/include/../hal/phydm/phydm_precomp.h:29,
                 from /usr/src/rtl88x2bu-5.6.1/include/hal_data.h:20,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_mlme.c:17:
/usr/src/rtl88x2bu-5.6.1/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from /usr/src/linux-headers-5.8.0-kali2-common/include/linux/filter.h:23,
                 from /usr/src/linux-headers-5.8.0-kali2-common/include/net/sock.h:59,
                 from /usr/src/linux-headers-5.8.0-kali2-common/include/linux/tcp.h:19,
                 from /usr/src/linux-headers-5.8.0-kali2-common/include/linux/ipv6.h:88,
                 from /usr/src/linux-headers-5.8.0-kali2-common/include/net/addrconf.h:50,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_service_linux.h:48,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.6.1/include/../hal/phydm/phydm_types.h:215,
                 from /usr/src/rtl88x2bu-5.6.1/include/../hal/phydm/phydm_precomp.h:29,
                 from /usr/src/rtl88x2bu-5.6.1/include/hal_data.h:20,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_mlme.c:17:
/usr/src/linux-headers-5.8.0-kali2-common/include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:79,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_io.c:47:
/usr/src/rtl88x2bu-5.6.1/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from /usr/src/linux-headers-5.8.0-kali2-common/include/linux/filter.h:23,
                 from /usr/src/linux-headers-5.8.0-kali2-common/include/net/sock.h:59,
                 from /usr/src/linux-headers-5.8.0-kali2-common/include/linux/tcp.h:19,
                 from /usr/src/linux-headers-5.8.0-kali2-common/include/linux/ipv6.h:88,
                 from /usr/src/linux-headers-5.8.0-kali2-common/include/net/addrconf.h:50,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_service_linux.h:48,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_io.c:47:
[... 77 lines cut ...]
                 from /usr/src/linux-headers-5.8.0-kali2-common/include/net/sock.h:59,
                 from /usr/src/linux-headers-5.8.0-kali2-common/include/linux/tcp.h:19,
                 from /usr/src/linux-headers-5.8.0-kali2-common/include/linux/ipv6.h:88,
                 from /usr/src/linux-headers-5.8.0-kali2-common/include/net/addrconf.h:50,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_service_linux.h:48,
                 from /usr/src/rtl88x2bu-5.6.1/include/osdep_service.h:50,
[... cut ...]
make[3]: *** [/usr/src/linux-headers-5.8.0-kali2-common/scripts/Makefile.build:286: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_cmd.o] Error 1
make[3]: *** [/usr/src/linux-headers-5.8.0-kali2-common/scripts/Makefile.build:286: /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_debug.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.8.0-kali2-common/Makefile:1764: /var/lib/dkms/rtl88x2bu/5.6.1/build] Error 2
make[1]: *** [/usr/src/linux-headers-5.8.0-kali2-common/Makefile:185: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-kali2-amd64'
make: *** [Makefile:2284: modules] Error 2
```

fpratbae, 2020-10-18:
errors also on 5.8.0-0.bpo.2-amd64 in Debian 10
[rtl88x2bu_5.6.1_make.log.txt](https://github.com/cilynx/rtl88x2bu/files/5397897/rtl88x2bu_5.6.1_make.log.txt)

limal, 2020-12-02:
This might help: https://dev.to/limal/install-rtl88x2ub-drivers-on-ubuntu-20-10-1ofp

CGarces, 2020-12-02:
Hi.

The error of redefinition of ‘struct sha256_state’ was solved by #58 
Are you sure that are using the branch 5.6.1_30362.20181109_COEX20180928-6a6a (the current default)?

limal, 2020-12-02:
I pulled the latest master which has the #58 already merged but the compilation problem still persists on Pop!_OS 20.10.

MaxG87, 2021-01-20:
Please try again with the most recent commits. Feel free to open a new issue if the problem still exists.

closed 2021-01-20 by MaxG87