cilynx/rtl88x2bu #90: sudo modprobe 88x2bu not working

digo-ca, 2020-10-16:
**command**: _sudo modprobe 88xbu_

**the execution never ends**

**uname -a:**
> _Linux U2COIN48402 5.4.0-51-generic #56-Ubuntu SMP Mon Oct 5 14:28:49 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux_

hsa1280, 2020-11-29:
I have the same issue, also the WiFi icon is not showing up on Ubuntu, maybe it is caused by this issue.

MaxG87, 2021-01-29:
Is this issue still relevant? Unfortunately, I do not know how to help here.

digo-ca, 2021-01-29:
Hello, I don't know what caused mine to work.  I believe the problem occurred when I performed the entire procedure with the wireless adapter connected.  I'm not sure if that was it, but when I did the procedure and only after I connected the adapter, it worked.

digo-ca, 2021-01-29:
But for it is no longer relevant, for me it could end without problems @MaxG87.

closed 2021-01-30 by MaxG87