cilynx/rtl88x2bu #96: can't compile on kernel 5.8.0-25-generic (Ubuntu 20.10)

OlaPaw, 2020-10-26:
I get the following error:
`make -j12 KERNELRELEASE=5.8.0-25-generic KVER=5.8.0-25-generic src=/usr/src/rtl88x2bu-5.6.1...(bad exit status: 2)`
`ERROR (dkms apport): binary package for rtl88x2bu: 5.6.1 not found
Error! Bad return status for module build on kernel: 5.8.0-25-generic (x86_64)`

looking into the log:
`DKMS make.log for rtl88x2bu-5.6.1 for kernel 5.8.0-25-generic (x86_64)
Mon 26 Oct 2020 05:06:09 PM EDT
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.8.0-25-generic/build M=/var/li
b/dkms/rtl88x2bu/5.6.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-25-generic'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_vht.o
In file included from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:79,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_debug.c:17:
/usr/src/rtl88x2bu-5.6.1/include/rtw_security.h:252:8: error: redefinition of ‘s
truct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:79,
                 from /usr/src/rtl88x2bu-5.6.1/core/rtw_ioctl_query.c:17:
/usr/src/rtl88x2bu-5.6.1/include/rtw_security.h:252:8: error: redefinition of ‘s
truct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23,
`

HughTebby, 2020-10-28:
#58 should help!

jackprogramsjp, 2020-10-28:
@HughTebby I can't find any solution in #58?

OlaPaw, 2020-10-29:
> #58 should help!

Thanks! That did work.

closed 2020-10-29 by OlaPaw