cilynx/rtl88x2bu #97: dmesg shows error

ssaid, 2020-10-28:
Hi, must i configure something to prevent this failure?
`[  922.243493] WARNING: CPU: 1 PID: 733 at /usr/src/rtl88x2bu-5.6.1/hal/hal_com.c:11513 rtw_lps_state_chk+0x40/0x46 [88x2bu]
[  922.243495] Modules linked in: fuse rfcomm nf_conntrack_netlink xfrm_user xfrm_algo xt_addrtype br_netfilter cmac bnep nft_chain_nat_ipv4 ipt_MASQUERADE nf_nat_ipv4 nf_nat xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_chain_route_ipv4 xt_CHECKSUM ipt_REJECT nf_reject_ipv4 nft_counter xt_tcpudp nft_compat tun bridge stp llc devlink nf_tables nfnetlink aufs(OE) overlay binfmt_misc intel_rapl nls_ascii nls_cp437 vfat fat x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel ath3k kvm btusb btrtl irqbypass btbcm snd_hda_codec_realtek btintel uvcvideo crct10dif_pclmul snd_hda_codec_hdmi snd_hda_codec_generic 88x2bu(OE) crc32_pclmul snd_hda_intel videobuf2_vmalloc bluetooth videobuf2_memops videobuf2_v4l2 snd_hda_codec videobuf2_common ghash_clmulni_intel cfg80211 intel_cstate snd_hda_core
[  922.243520]  videodev snd_hwdep intel_uncore drbg efi_pstore intel_rapl_perf media ansi_cprng snd_pcm ecdh_generic sg rtsx_pci_ms joydev iTCO_wdt memstick snd_timer ideapad_laptop mei_me sparse_keymap iTCO_vendor_support snd efivars rfkill mei pcspkr soundcore serio_raw wmi ac pcc_cpufreq battery evdev efivarfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 fscrypto ecb btrfs zstd_decompress zstd_compress xxhash raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c crc32c_generic raid1 raid0 multipath linear md_mod hid_generic usbhid hid sr_mod sd_mod cdrom i915 crc32c_intel rtsx_pci_sdmmc mmc_core i2c_algo_bit drm_kms_helper ahci aesni_intel ehci_pci libahci drm ehci_hcd aes_x86_64 crypto_simd cryptd xhci_pci glue_helper xhci_hcd psmouse i2c_i801 libata
[  922.243555]  r8169 usbcore realtek libphy scsi_mod lpc_ich rtsx_pci mfd_core thermal usb_common video button
[  922.243564] CPU: 1 PID: 733 Comm: RTW_CMD_THREAD Tainted: G        W  OE     4.19.0-12-amd64 #1 Debian 4.19.152-1
[  922.243565] Hardware name: LENOVO 80F3/Lenovo B40-70, BIOS 9DCN29WW(V2.09) 11/21/2014
[  922.243623] RIP: 0010:rtw_lps_state_chk+0x40/0x46 [88x2bu]
[  922.243625] Code: 06 00 00 48 89 ef e8 1b 64 f9 ff 84 c0 79 1d bf 01 00 00 00 e8 1a 82 fd ff 80 eb 01 75 e0 48 c7 c7 30 11 e9 c0 e8 1b 67 91 ee <0f> 0b 5b 5d c3 c3 0f 1f 44 00 00 55 53 48 83 ec 10 48 89 d5 65 48
[  922.243626] RSP: 0018:ffffa1bdc1847d00 EFLAGS: 00010246
[  922.243627] RAX: 0000000000000024 RBX: 0000000000000000 RCX: 0000000000000000
[  922.243629] RDX: 0000000000000000 RSI: ffff9021170566b8 RDI: ffff9021170566b8
[  922.243630] RBP: ffffa1bdc12a1000 R08: 000000000000048e R09: 0000000000000004
[  922.243631] R10: 0000000000000000 R11: 0000000000000001 R12: ffffa1bdc1847db4
[  922.243631] R13: ffffa1bdc1351000 R14: 0000000000000001 R15: ffffffffc0d56cbe
[  922.243633] FS:  0000000000000000(0000) GS:ffff902117040000(0000) knlGS:0000000000000000
[  922.243634] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  922.243635] CR2: 00000eaf08db4000 CR3: 000000019180a004 CR4: 00000000001606e0
[  922.243637] Call Trace:
[  922.243695]  SetHwReg+0x6c1/0x6f5 [88x2bu]
[  922.243750]  rtl8822b_sethwreg+0xdc0/0xdee [88x2bu]
[  922.243804]  sethwreg+0x127/0x12e [88x2bu]
[  922.243859]  rtw_hal_set_hwreg+0x15/0x16 [88x2bu]
[  922.243901]  rtw_set_ps_mode+0xe4/0x210 [88x2bu]
[  922.243942]  LPS_Leave+0xc6/0xcd [88x2bu]
[  922.243996]  rtl8822b_phy_haldm_watchdog+0x175/0x19b [88x2bu]
[  922.244051]  rtw_hal_dm_watchdog+0x1d/0x1f [88x2bu]
[  922.244076]  rtw_dynamic_chk_wk_hdl+0x33/0x4d [88x2bu]
[  922.244102]  rtw_drvextra_cmd_hdl+0x2aa/0x48a [88x2bu]
[  922.244127]  rtw_cmd_thread+0x2d6/0x447 [88x2bu]
[  922.244151]  ? rtw_stop_cmd_thread+0x39/0x39 [88x2bu]
[  922.244155]  kthread+0x112/0x130
[  922.244157]  ? kthread_bind+0x30/0x30
[  922.244160]  ret_from_fork+0x35/0x40
[  922.244163] ---[ end trace 1db2dbf509786ee8 ]---
`

ssaid, 2020-10-28:
> [  922.243493] WARNING: CPU: 1 PID: 733 at /usr/src/rtl88x2bu-5.6.1/hal/hal_com.c:11513 rtw_lps_state_chk+0x40/0x46 [88x2bu]
[  922.243495] Modules linked in: fuse rfcomm nf_conntrack_netlink xfrm_user xfrm_algo xt_addrtype br_netfilter cmac bnep nft_chain_nat_ipv4 ipt_MASQUERADE nf_nat_ipv4 nf_nat xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_chain_route_ipv4 xt_CHECKSUM ipt_REJECT nf_reject_ipv4 nft_counter xt_tcpudp nft_compat tun bridge stp llc devlink nf_tables nfnetlink aufs(OE) overlay binfmt_misc intel_rapl nls_ascii nls_cp437 vfat fat x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel ath3k kvm btusb btrtl irqbypass btbcm snd_hda_codec_realtek btintel uvcvideo crct10dif_pclmul snd_hda_codec_hdmi snd_hda_codec_generic 88x2bu(OE) crc32_pclmul snd_hda_intel videobuf2_vmalloc bluetooth videobuf2_memops videobuf2_v4l2 snd_hda_codec videobuf2_common ghash_clmulni_intel cfg80211 intel_cstate snd_hda_core
[  922.243520]  videodev snd_hwdep intel_uncore drbg efi_pstore intel_rapl_perf media ansi_cprng snd_pcm ecdh_generic sg rtsx_pci_ms joydev iTCO_wdt memstick snd_timer ideapad_laptop mei_me sparse_keymap iTCO_vendor_support snd efivars rfkill mei pcspkr soundcore serio_raw wmi ac pcc_cpufreq battery evdev efivarfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 fscrypto ecb btrfs zstd_decompress zstd_compress xxhash raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c crc32c_generic raid1 raid0 multipath linear md_mod hid_generic usbhid hid sr_mod sd_mod cdrom i915 crc32c_intel rtsx_pci_sdmmc mmc_core i2c_algo_bit drm_kms_helper ahci aesni_intel ehci_pci libahci drm ehci_hcd aes_x86_64 crypto_simd cryptd xhci_pci glue_helper xhci_hcd psmouse i2c_i801 libata
[  922.243555]  r8169 usbcore realtek libphy scsi_mod lpc_ich rtsx_pci mfd_core thermal usb_common video button
[  922.243564] CPU: 1 PID: 733 Comm: RTW_CMD_THREAD Tainted: G        W  OE     4.19.0-12-amd64 #1 Debian 4.19.152-1
[  922.243565] Hardware name: LENOVO 80F3/Lenovo B40-70, BIOS 9DCN29WW(V2.09) 11/21/2014
[  922.243623] RIP: 0010:rtw_lps_state_chk+0x40/0x46 [88x2bu]
[  922.243625] Code: 06 00 00 48 89 ef e8 1b 64 f9 ff 84 c0 79 1d bf 01 00 00 00 e8 1a 82 fd ff 80 eb 01 75 e0 48 c7 c7 30 11 e9 c0 e8 1b 67 91 ee <0f> 0b 5b 5d c3 c3 0f 1f 44 00 00 55 53 48 83 ec 10 48 89 d5 65 48
[  922.243626] RSP: 0018:ffffa1bdc1847d00 EFLAGS: 00010246
[  922.243627] RAX: 0000000000000024 RBX: 0000000000000000 RCX: 0000000000000000
[  922.243629] RDX: 0000000000000000 RSI: ffff9021170566b8 RDI: ffff9021170566b8
[  922.243630] RBP: ffffa1bdc12a1000 R08: 000000000000048e R09: 0000000000000004
[  922.243631] R10: 0000000000000000 R11: 0000000000000001 R12: ffffa1bdc1847db4
[  922.243631] R13: ffffa1bdc1351000 R14: 0000000000000001 R15: ffffffffc0d56cbe
[  922.243633] FS:  0000000000000000(0000) GS:ffff902117040000(0000) knlGS:0000000000000000
[  922.243634] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  922.243635] CR2: 00000eaf08db4000 CR3: 000000019180a004 CR4: 00000000001606e0
[  922.243637] Call Trace:
[  922.243695]  SetHwReg+0x6c1/0x6f5 [88x2bu]
[  922.243750]  rtl8822b_sethwreg+0xdc0/0xdee [88x2bu]
[  922.243804]  sethwreg+0x127/0x12e [88x2bu]
[  922.243859]  rtw_hal_set_hwreg+0x15/0x16 [88x2bu]
[  922.243901]  rtw_set_ps_mode+0xe4/0x210 [88x2bu]
[  922.243942]  LPS_Leave+0xc6/0xcd [88x2bu]
[  922.243996]  rtl8822b_phy_haldm_watchdog+0x175/0x19b [88x2bu]
[  922.244051]  rtw_hal_dm_watchdog+0x1d/0x1f [88x2bu]
[  922.244076]  rtw_dynamic_chk_wk_hdl+0x33/0x4d [88x2bu]
[  922.244102]  rtw_drvextra_cmd_hdl+0x2aa/0x48a [88x2bu]
[  922.244127]  rtw_cmd_thread+0x2d6/0x447 [88x2bu]
[  922.244151]  ? rtw_stop_cmd_thread+0x39/0x39 [88x2bu]
[  922.244155]  kthread+0x112/0x130
[  922.244157]  ? kthread_bind+0x30/0x30
[  922.244160]  ret_from_fork+0x35/0x40
[  922.244163] ---[ end trace 1db2dbf509786ee8 ]---

MaxG87, 2020-12-25:
Can you please provide a bit of context? What action triggered that DKMS problem? What are the details of your system?

closed 2021-01-29 by MaxG87