cilynx/rtl88x2bu #98: How do I get this driver to work on 64bit raspberry pi OS?

ThomasHabets, 2020-10-29:
I'm using [the 64bit Beta of raspberry pi OS](https://www.raspberrypi.org/forums/viewtopic.php?t=275370).

I've changed `ARCH` in the Makefile to be `arm64`. But then I had to add this to a c file:

```
unsigned long __stack_chk_guard;
void __stack_chk_guard_setup(void)
{
     __stack_chk_guard = 0xBAAAAAAD;
}

void __stack_chk_fail(void)
```

Then it works  for one of my cards.

The other card, with same USB ID of 0bda:b812 but different formfactor, it does not. dmesg shows:

```
usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
````

But there is no cable. It's plugged directly into the Pi.

I'll try it on a windows machine, to see if it's broken HW.

PieGuy314, 2020-11-03:
Same here. Got it to compile and run on aarch64 Raspberry Pi 4 by patching the Makefile.

I've created a GitHub repo that details the process I used... [https://github.com/PieGuy314/RTL88x2BU-RPi4-arm64-Driver-Patch](https://github.com/PieGuy314/RTL88x2BU-RPi4-arm64-Driver-Patch)

MaxG87, 2021-01-01:
Thank you for the useful pointers. Would it be possible for you to provide a Pull Request for us to apply?

PieGuy314, 2021-01-02:
> Thank you for the useful pointers. Would it be possible for you to provide a Pull Request for us to apply?

Hi,

Last time I checked, the latest Raspberry Pi headers hadn't been patched properly. As a consequence, you couldn't compile the driver against that specific kernel. However, there's a newer kernel out now. I will however, try a vanilla install of DietPi 64-bit tomorrow on my Pi-4B to see that's occurring.

PieGuy314, 2021-01-08:
A newer version of raspberrypi-kernel dropped today (1.20200108-1, with kernel 5.4.83) - Headers are once again broken for 64-bit RTL88x2BU driver builds.

I've reverted to 32-bit on the one system that requires an externally compiled driver.

closed 2021-06-03 by MaxG87