cilynx/rtl88x2bu #99: Kernel 5.10 RC2: Error! Bad return status for module build on kernel: 5.10.0-051000rc2-generic (x86_64)

TheJags, 2020-11-02:
Building module fails while trying to install Kernel 5.10 RC2 in Ubuntu MATE Groovy 20.10  with the following error message:

````
Building module:
cleaning build area...
make -j12 KERNELRELEASE=5.10.0-051000rc2-generic KVER=5.10.0-051000rc2-generic src=/usr/src/rtl88x2bu-5.6.1......(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.0-051000rc2-generic (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.6.1/build/make.log for more information.
   ...done.
````
/var/lib/dkms/rtl88x2bu/5.6.1/build/make.log

````
DKMS make.log for rtl88x2bu-5.6.1 for kernel 5.10.0-051000rc2-generic (x86_64)

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.0-051000rc2-generic/build M=/var/lib/dkms/rtl88x2bu/5.6.1/build  modules

make[1]: Entering directory '/usr/src/linux-headers-5.10.0-051000rc2-generic'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/wifi_regd.o

/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c: In function ‘isFileReadable’:
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2203:11: error: implicit declaration of function ‘get_fs’; did you mean ‘get_sa’? [-Werror=implicit-function-declaration]
 2203 |   oldfs = get_fs();
      |           ^~~~~~
      |           get_sa

/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2203:11: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2205:3: error: implicit declaration of function ‘set_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
 2205 |   set_fs(KERNEL_DS);
      |   ^~~~~~
      |   sget_fc

/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2205:10: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2205 |   set_fs(KERNEL_DS);
      |          ^~~~~~~~~
      |          KERNFS_NS

/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2205:10: note: each undeclared identifier is reported only once for each function it appears in
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c: In function ‘retriveFromFile’:
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2245:12: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
 2245 |    oldfs = get_fs();
      |            ^~~~~~

/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2247:11: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2247 |    set_fs(KERNEL_DS);
      |           ^~~~~~~~~
      |           KERNFS_NS

/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c: In function ‘storeToFile’:
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2284:12: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
 2284 |    oldfs = get_fs();
      |            ^~~~~~

/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2286:11: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2286 |    set_fs(KERNEL_DS);
      |           ^~~~~~~~~
      |           KERNFS_NS

cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:283: /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/osdep_service.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1799: /var/lib/dkms/rtl88x2bu/5.6.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-051000rc2-generic'
make: *** [Makefile:2284: modules] Error 2
````

MaxG87, 2020-11-12:
Which revision are your using to compile? I think some issues with `get_fs()` were fixed previously.

CGarces, 2020-11-12:
Hi all.

The code must be adapted to this kernel modification
https://github.com/torvalds/linux/commit/f56e65dff6ad52395ef45738799b4fb70ff43376

Please use this patch as reference and give me some feedback.
https://github.com/Mange/rtl8192eu-linux-driver/pull/201/

TheJags, 2020-11-16:
> Which revision are your using to compile? I think some issues with `get_fs()` were fixed previously.

That was **with** 5.8 compatibility patch.

TheJags, 2020-11-16:
> Hi all.
> 
> The code must be adapted to this kernel modification
> [torvalds/linux@f56e65d](https://github.com/torvalds/linux/commit/f56e65dff6ad52395ef45738799b4fb70ff43376)
> 
> Please use this patch as reference and give me some feedback.
> [Mange/rtl8192eu-linux-driver#201](https://github.com/Mange/rtl8192eu-linux-driver/pull/201)

Is this how I should be applying the patch? Because I'm getting the following errors.

Just wanna make sure as this is a repo for RTL88x2BU and the patch is from RTL8192EU, correct?
````
git clone https://github.com/cilynx/rtl88x2bu.git

cd rtl88x2bu

wget https://github.com/Mange/rtl8192eu-linux-driver/pull/201.patch

--2020-11-16 22:00:21--  https://github.com/Mange/rtl8192eu-linux-driver/pull/201.patch
Resolving github.com (github.com)... 13.234.210.38
Connecting to github.com (github.com)|13.234.210.38|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://patch-diff.githubusercontent.com/raw/Mange/rtl8192eu-linux-driver/pull/201.patch [following]
--2020-11-16 22:00:21--  https://patch-diff.githubusercontent.com/raw/Mange/rtl8192eu-linux-driver/pull/201.patch
Resolving patch-diff.githubusercontent.com (patch-diff.githubusercontent.com)... 13.234.210.38
Connecting to patch-diff.githubusercontent.com (patch-diff.githubusercontent.com)|13.234.210.38|:443... connected.
HTTP request sent, awaiting response... 200 OK
Cookie coming from patch-diff.githubusercontent.com attempted to set domain to github.com
Length: unspecified [text/plain]
Saving to: ‘201.patch’

201.patch                                           [ <=>                                                                                                  ]   5.96K  --.-KB/s    in 0s      

2020-11-16 22:00:22 (30.4 MB/s) - ‘201.patch’ saved [6100]

git apply 201.patch

error: patch failed: os_dep/osdep_service.c:2193
error: os_dep/osdep_service.c: patch does not apply
````

CGarces, 2020-11-16:
> Is this how I should be applying the patch? Because I'm getting the following errors.
> 
> Just wanna make sure as this is a repo for RTL88x2BU and the patch is from RTL8192EU, correct?

As I said, use the patch _as reference_ of what is needed to patch the driver.
You must create your own patch (and send a PR).

TheJags, 2020-11-16:
> As I said, use the patch _as reference_ of what is needed to patch the driver.
> You must create your own patch (and send a PR).

Oh ok. It was a long day, and I totally missed it. Thanks alot.

MaxG87, 2020-11-23:
I'm sorry for answering so late.

The 5.8 compatibility patches were merged in both relevant branches. Maybe you try out the new branch tips for a fresh start.

Regarding you applying the patch using `git apply`:

1. The end of the log states your patch could not be applied. Thus, you were compiling your code without the expected changes.
2. In general I find it more convenient to add a new remote and use `git merge` or `git cherry-pick`. This gives you the opportunity to inspect the differences better and to resolve conflicts using `git mergetool`.

desegura, 2020-12-08:
Hi anyone that make it run, can please upload the patched driver for the rtl88x2bu ?

CGarces, 2020-12-09:
@desegura 
The patch at Mange/rtl8192eu-linux-driver#201 should be very easy to adapt to this repository.

MaxG87, 2020-12-11:
@CGarces Thank you very much for the link. Your patch set was a great help.

I created a fix and a pull request. You can test it using the version of https://github.com/MaxG87/rtl88x2bu.

MaxG87, 2020-12-16:
Closed by #106.

closed 2020-12-16 by MaxG87