cilynx/rtl88x2bu #100: Fails to build on Ubuntu 20.10

mortenvp, 2020-11-04:
Hi guys, 
I'm having a problem building on Ubuntu 20.10. I think the issue/fix is the same as described here:
https://github.com/aircrack-ng/rtl8188eus/pull/77

Here is the log from the build output:

```
cat /var/lib/dkms/rtl88x2bu/5.3.1/build/make.log                                                                              (base) 
DKMS make.log for rtl88x2bu-5.3.1 for kernel 5.8.0-26-generic (x86_64)
Mi 4. Nov 10:12:23 CET 2020
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.8.0-26-generic/build M=/var/lib/dkms/rtl88x2bu/5.3.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-26-generic'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_sta_mgt.o
In file included from /usr/src/rtl88x2bu-5.3.1/include/drv_types.h:78,
                 from /usr/src/rtl88x2bu-5.3.1/core/rtw_security.c:17:
/usr/src/rtl88x2bu-5.3.1/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service_linux.h:48,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.3.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.3.1/core/rtw_security.c:17:
./include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from /usr/src/rtl88x2bu-5.3.1/include/drv_types.h:78,
                 from /usr/src/rtl88x2bu-5.3.1/core/rtw_io.c:47:
/usr/src/rtl88x2bu-5.3.1/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service_linux.h:48,
                 from /usr/src/rtl88x2bu-5.3.1/include/osdep_service.h:50,
                 from /usr/src/rtl88x2bu-5.3.1/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-5.3.1/core/rtw_io.c:47:
./include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from /usr/src/rtl88x2bu-5.3.1/include/drv_types.h:78,
                 from /usr/src/rtl88x2bu-5.3.1/core/rtw_cmd.c:17:
/usr/src/rtl88x2bu-5.3.1/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
[... 255 lines cut ...]
make[2]: *** [scripts/Makefile.build:290: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ieee80211.o] Error 1
make[2]: *** [scripts/Makefile.build:290: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_io.o] Error 1
make[2]: *** [scripts/Makefile.build:290: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rf.o] Error 1
make[2]: *** [scripts/Makefile.build:290: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_wlan_util.o] Error 1
make[2]: *** [scripts/Makefile.build:290: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_recv.o] Error 1
make[2]: *** [scripts/Makefile.build:290: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ioctl_set.o] Error 1
make[2]: *** [scripts/Makefile.build:290: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mi.o] Error 1
make[2]: *** [scripts/Makefile.build:290: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_pwrctrl.o] Error 1
make[2]: *** [scripts/Makefile.build:290: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_vht.o] Error 1
make[2]: *** [scripts/Makefile.build:290: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_cmd.o] Error 1
make[2]: *** [scripts/Makefile.build:290: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mlme.o] Error 1
make[2]: *** [scripts/Makefile.build:290: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_debug.o] Error 1
/usr/src/rtl88x2bu-5.3.1/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/usr/src/rtl88x2bu-5.3.1/core/rtw_mlme_ext.c:1828:17: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1828 |    ptable->func = &OnAuthClient;
      |    ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/usr/src/rtl88x2bu-5.3.1/core/rtw_mlme_ext.c:1830:2: note: here
 1830 |  case WIFI_ASSOCREQ:
      |  ^~~~
make[2]: *** [scripts/Makefile.build:290: /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mlme_ext.o] Error 1
make[1]: *** [Makefile:1780: /var/lib/dkms/rtl88x2bu/5.3.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-26-generic'
make: *** [Makefile:2011: modules] Error 2

```

alphazilla, 2020-11-06:
Same issues after upgrading to **Pop!_OS 20.10** my wireless adaptor stopped working.

limal, 2020-12-02:
I had the same issue on Pop!_OS 20.10 and what helped was using https://github.com/morrownr/88x2bu instead.

I wrote an small article about it here: https://dev.to/limal/install-rtl88x2ub-drivers-on-ubuntu-20-10-1ofp

MaxG87, 2021-01-18:
What Kernel are you trying to compile for? Could you please retry with the most recent version of this project?

I recently compiled the 5.6.1.6_... branch for v5.10, so I wonder a bit why it should not work for v5.8.

mortenvp, 2021-01-19:
I've switched WiFi adapter so can't test it anymore. I will close - feel free to re-open if somebody is able to reproduce.

closed 2021-01-19 by mortenvp