cilynx/rtl88x2bu #102: [Question] How to update driver when code is updated?

HTV04, 2020-11-21:
This less of an issue and more of a dumb question that I have.

When an update to the code is pushed, what is the "proper" way to update the code? Do I just follow the steps to install the driver with DKMS again, or do I run "dkms remove [module]" then rerun the steps to install it again? I'm new to Ubuntu (and Linux in general) and I really don't want to screw anything up. Thanks!

pfille, 2021-01-08:
Might be a little late and I'm not sure if this is the "proper" way, but here is what worked for me after the module failed building due to system updates:
I'm currently on Ubuntu 20.04 with kernel 5.8.0-34-generic and here is the original error message:
```
/etc/kernel/postinst.d/dkms:
 * dkms: running auto installation service for kernel 5.8.0-36-generic

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.8.0-36-generic KVER=5.8.0-36-generic src=/usr/src/rtl88x2bu-5.6.1...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl88x2bu: 5.6.1 not found
Error! Bad return status for module build on kernel: 5.8.0-36-generic (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.6.1/build/make.log for more information.
```
Navigate to the source directory which is referenced in the error message:
`$cd /usr/src/rtl88x2bu-5.6.1` (in my case)

The directory should be a git repo (maybe check with `$git status`) and pull the changes from the repository origin to update the driver module:
`$sudo git pull origin`

Then manually initiate the dkms auto installation service:
`$sudo dkms autoinstall`

Start wifi adapter:
`$sudo modprobe 88x2bu`

MaxG87, 2021-01-08:
It should be posible to run the deploy script of 5.6.1.6_35492.20191025_COEX20180928-6a6a for updating the driver.
However, I did not try this out for some time.

TheJags, 2021-01-08:
@pfille @MaxG87 

I just try to update drivers as described by 'pfille' and got the following output. Would any of you like to modify the command or no?

Thanks alot in advance.

Note: Though the GitHub repo I'm using at the moment is different but I think the process would be the same.

````
$ sudo git pull origin

warning: Pulling without specifying how to reconcile divergent branches is
discouraged. You can squelch this message by running one of the following
commands sometime before your next pull:

  git config pull.rebase false  # merge (the default strategy)
  git config pull.rebase true   # rebase
  git config pull.ff only       # fast-forward only

You can replace "git config" with "git config --global" to set a default
preference for all repositories. You can also pass --rebase, --no-rebase,
or --ff-only on the command line to override the configured default per
invocation.

remote: Enumerating objects: 5, done.
remote: Counting objects: 100% (5/5), done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 3 (delta 2), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (3/3), 652 bytes | 130.00 KiB/s, done.

From https://github.com/RinCat/RTL88x2BU-Linux-Driver
   ad2d09a..afc917d  master     -> origin/master

Updating ad2d09a..afc917d
Fast-forward
 README.md | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)
````

pfille, 2021-01-08:
@TheJags
Disclaimer: Maybe it would be a good idea to see if someone with more experience chimes in to give advice and be cautious to proceed for now.

After having read through [here](https://stackoverflow.com/questions/62653114/how-to-deal-with-this-git-warning-pulling-without-specifying-how-to-reconcile) I realized that I didn't encounter the same warning upon `git pull` as you did, since my git version is older than 2.27. From what I understand, your repository should contain the most recent code changes now and you should be fine, but your git commit history might be different from what you expect as `git pull` also performs a merge in the background.

Have you tried running `$ sudo dkms autoinstall` already and was is successful?

TheJags, 2021-01-08:
@pfille 

> From what I understand, your local repository should contain the most recent code changes from https://github.com/RinCat/RTL88x2BU-Linux-Driver now and you should be fine, but your git commit history might be different from what you expect as `git pull` also performs a merge in the background.

I guess that's okay.

> Have you tried running `$ sudo dkms autoinstall` already and was it successful?

In my case drivers were working already... but there were 1 or 2 commit AFTER I "clean-install" few weeks ago... so I just thought to try and update as you've described.

I guess I will come to know when next kernel change requires me to update the drivers.

Thanks alot.

HTV04, 2021-01-09:
@pfille Sorry for the late reply, but thanks so much! I guess it's safe to close this now.

closed 2021-01-09 by HTV04