cilynx/rtl88x2bu #103: Misses some networks

patrakov, 2020-11-23:
I have a laptop with a buit-in WiFi network card which is unfortunately linited to 2.4 GHz only:

```
01:00.0 Network controller [0280]: Intel Corporation Wireless 7260 [8086:08b2] (rev 6b)
```

And a TP-Link Archer T4U ver.3.2, which is supported by your driver:

```
Bus 003 Device 004: ID 2357:0115 TP-Link Archer T4U ver.3
```

The kernel is 5.9.9-arch1-1, the default from Arch Linux. The driver is built from git commit 439c7dbd9e43acd91d5704a50b68647716c52505. The problem is that there are some access points (including my own!) in the 2.4 GHz band that can be seen by the onboard WiFi adapter, but not the USB adapter.

The problem does not exist with the driver from https://github.com/ulli-kroll/rtw88-usb but that driver crashes often on big uploads.

Scan results (iw dev XXX scan) from the onboard adapter, the USB adapter with the 88x2bu driver, and from the USb adapter with the rtw88-usb driver, are attached below.

[wlp1s0-onboard-scan.txt](https://github.com/cilynx/rtl88x2bu/files/5582199/wlp1s0-onboard-scan.txt)
[wlp0s20u2-88x2bu-scan.txt](https://github.com/cilynx/rtl88x2bu/files/5582203/wlp0s20u2-88x2bu-scan.txt)
[wlp0s20u2-rtw88-usb-scan.txt](https://github.com/cilynx/rtl88x2bu/files/5582221/wlp0s20u2-rtw88-usb-scan.txt)

Note that "t503" and "WiFiLab" BSSIDs are in fact provided by the same access point (Netgear R7800 running an OpenWRT snapshot), but with different authentication settings. And only one of them is seen by the adapter.

Also, it is not only "t503" which is missing, so you cannot blame OpenWRT.

patrakov, 2020-12-03:
With kernel 5.9.10-arch1-1, no such issues. I will watch for some more time and close the issue if it doesn't reappear.

[wlp0s20u2-88x2bu-scan.txt](https://github.com/cilynx/rtl88x2bu/files/5637274/wlp0s20u2-88x2bu-scan.txt)

MaxG87, 2020-12-25:
Thank you for the bug report. Do you think you extract a patch from https://github.com/ulli-kroll/rtw88-usb? I would love to apply it here too.

patrakov, 2020-12-25:
No, that driver is not based on Realtek-provided sources.

MaxG87, 2020-12-28:
Fixing such subtle problems is beyond my capabilities. I am very sorry for this.

Maybe a more recent version of the driver would fix that bug. Maybe you try out https://github.com/ulli-kroll/rtw88-usb. I would love to hear your report.

As soon as I find an primary source for a more recent version, I will create a new branch for that version.

closed 2020-12-28 by MaxG87