cilynx/rtl88x2bu #104: everything works until the modprobe (Fedora, kernel 5.9.10)

ICRaptor, 2020-11-30:
keep getting `modprobe: ERROR: could not insert '88x2bu': Exec format error` after doing `make` `sudo make install` 

same error happens when running the DKMS installation or using the `build.sh` script

MaxG87, 2020-12-25:
Have you managed to resolve the issue?

Maybe using the 5.6.1.6 branch might help. Could you please try that out and report back?

ICRaptor, 2020-12-25:
ended up purchasing an Intel AX200 PCIe x1 card to at least get up and running, eventually I decided I would reinstall the RTL88x2BU driver and it ran without issue -- my assumption is it was broken for some reason and after a Fedora update, it worked again. 

No clue as to why but I have it working now.

Also, using the 5.6.1 and 5.6.1.6 have worked for me in the past, I had originally used the 5.6.1.6 and gotten the issue stated above. Switching to 5.6.1 didn't have an impact

MaxG87, 2020-12-25:
Good to hear. Lets hope it was some glitch in the combination of driver and Fedora.

closed 2020-12-25 by MaxG87