cilynx/rtl88x2bu #108: Fails to compile on Solus

Tr0ubledC0de, 2020-12-28:
It errors around building. The errors seems to be coming from "osdep_service.c" in the "os_dep" folder.

Here are my logs (ignore Tomomi; I was trying to build it another way but it all leads to the same issue as running "make"):
Copyright (c) 2019-2020, Alexander Vereeken All rights reserved.
Welcome to Tomomi Live have fun!
Downloading driver
Cloning into 'rtl88x2bu'...
remote: Enumerating objects: 76, done.
remote: Counting objects: 100% (76/76), done.
remote: Compressing objects: 100% (67/67), done.
remote: Total 1217 (delta 20), reused 26 (delta 9), pack-reused 1141
Receiving objects: 100% (1217/1217), 5.06 MiB | 241.00 KiB/s, done.
Resolving deltas: 100% (594/594), done.
Building driver
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.2-164.current/build M=/home/root/Tomomi/rtl88x2bu  modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.2-164.current'
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_cmd.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_security.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_debug.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_io.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_ioctl_query.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_ioctl_set.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_ieee80211.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_mlme.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_mlme_ext.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_mi.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_wlan_util.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_vht.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_pwrctrl.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_rf.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_chplan.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_recv.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_sta_mgt.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_ap.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/mesh/rtw_mesh.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/mesh/rtw_mesh_pathtbl.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/mesh/rtw_mesh_hwmp.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_xmit.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_p2p.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_rson.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_tdls.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_br_ext.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_iol.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_sreset.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_btcoex_wifionly.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_btcoex.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_beamforming.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_odm.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_rm.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/rtw_rm_fsm.o
CC [M]  /home/root/Tomomi/rtl88x2bu/core/efuse/rtw_efuse.o
CC [M]  /home/root/Tomomi/rtl88x2bu/os_dep/osdep_service.o
/home/root/Tomomi/rtl88x2bu/os_dep/osdep_service.c: In function ‘isFileReadable’:
/home/root/Tomomi/rtl88x2bu/os_dep/osdep_service.c:2203:11: error: implicit declaration of function ‘get_fs’; did you mean ‘get_sa’? [-Werror=implicit-function-declaration]
2203 |   oldfs = get_fs();
|           ^~~~~~
|           get_sa
/home/root/Tomomi/rtl88x2bu/os_dep/osdep_service.c:2203:11: error: incompatible types when assigning to type ‘mm_segment_t’ {aka ‘struct ’} from type ‘int’
/home/root/Tomomi/rtl88x2bu/os_dep/osdep_service.c:2205:3: error: implicit declaration of function ‘set_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
2205 |   set_fs(KERNEL_DS);
|   ^~~~~~
|   sget_fc
/home/root/Tomomi/rtl88x2bu/os_dep/osdep_service.c:2205:10: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
2205 |   set_fs(KERNEL_DS);
|          ^~~~~~~~~
|          KERNFS_NS
/home/root/Tomomi/rtl88x2bu/os_dep/osdep_service.c:2205:10: note: each undeclared identifier is reported only once for each function it appears in
/home/root/Tomomi/rtl88x2bu/os_dep/osdep_service.c: In function ‘retriveFromFile’:
/home/root/Tomomi/rtl88x2bu/os_dep/osdep_service.c:2245:12: error: incompatible types when assigning to type ‘mm_segment_t’ {aka ‘struct ’} from type ‘int’
2245 |    oldfs = get_fs();
|            ^~~~~~
/home/root/Tomomi/rtl88x2bu/os_dep/osdep_service.c:2247:11: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
2247 |    set_fs(KERNEL_DS);
|           ^~~~~~~~~
|           KERNFS_NS
/home/root/Tomomi/rtl88x2bu/os_dep/osdep_service.c: In function ‘storeToFile’:
/home/root/Tomomi/rtl88x2bu/os_dep/osdep_service.c:2284:12: error: incompatible types when assigning to type ‘mm_segment_t’ {aka ‘struct ’} from type ‘int’
2284 |    oldfs = get_fs();
|            ^~~~~~
/home/root/Tomomi/rtl88x2bu/os_dep/osdep_service.c:2286:11: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
2286 |    set_fs(KERNEL_DS);
|           ^~~~~~~~~
|           KERNFS_NS
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:279: /home/root/Tomomi/rtl88x2bu/os_dep/osdep_service.o] Error 1
make[1]: *** [Makefile:1805: /home/root/Tomomi/rtl88x2bu] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.2-164.current'
make: *** [Makefile:2284: modules] Error 2
Installing driver
install -p -m 644 88x2bu.ko  /lib/modules/5.10.2-164.current/kernel/drivers/net/wireless/
install: cannot stat '88x2bu.ko': No such file or directory
make: *** [Makefile:2290: install] Error 1
modprobe: FATAL: Module 88x2bu not found in directory /lib/modules/5.10.2-164.current
Done :) You can now use your wifi adapter!

CGarces, 2020-12-28:
Hi.

Default branch not has the 5.10 patch
You have two options:

- use 5.6.1.6_35492.20191025_COEX20180928-6a6a branch.
- use 5.6.1_30362.20181109_COEX20180928-6a6a and cherry-pick comit https://github.com/cilynx/rtl88x2bu/commit/8b3babea2ccf874e0ce2e9ec1e1ca91efd8e826f, I don't know if will apply in a clean way.

Bonus: send a PR to default branch.

MaxG87, 2021-01-20:
Fixed by #121 .

closed 2021-01-20 by MaxG87