cilynx/rtl88x2bu #113: Cannot compile on arm64 debian linux kernel 5.6.

IModThings, 2021-01-07:
DOES compile and works perfectly on Fedora 33 kernel 5.9 on a x86-64 processor however I need this for a rock64 board NAS running debian linux kernel 5.6 on an ARM Cortex A53. While this may be an semi-unusual use-case the Makefile does have options for other RK processors so I assume it can't hurt to ask for RK3328 build flag as this is a fairly popular SBC.

CGarces, 2021-01-07:
Please provide a log with the error and the options used to call the make command.

IModThings, 2021-01-07:
```
DKMS make.log for rtl88x2bu-5.6.1 for kernel 5.6.0-1138-ayufan-g2d8a7fc1c7ca (aarch64)
Thu 07 Jan 2021 11:20:03 AM UTC
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.6.0-1138-ayufan-g2d8a7fc1c7ca/build M=/var/lib/dkms/rtl88x2bu/5.6.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.6.0-1138-ayufan-g2d8a7fc1c7ca'
Makefile:653: arch/arm/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/arm/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-5.6.0-1138-ayufan-g2d8a7fc1c7ca'
make: *** [Makefile:2284: modules] Error 2

Following the raspberry pi instructions. Testing the other RK chip options and some others also did not work.

I can provide the detailed log as well but it is mostly a long list of various missing files such as 'arch/arm/Makefile'.

CGarces, 2021-01-07:
I'm not sure but, if your kernel is aarch64 maybe you must use `make ARCH=arm64`?

IModThings, 2021-01-07:
Yep... *facepalm* I really should learn how compiling stuff works... Could have saved many hours...
Thank you very much.

I can confirm it works on a rock64 1gb SBC running debian, linux kernel 5.6.

closed 2021-01-07 by IModThings