cilynx/rtl88x2bu #114: ERROR: could not insert '88x2bu': Unknown symbol in module, or unknown parameter (see dmesg)

hdnh2006, 2021-01-07:
Hi, I followed all the steps, but when I try the final one (`sudo modprobe 88x2bu`) I get the following error:

`ERROR: could not insert '88x2bu': Unknown symbol in module, or unknown parameter (see dmesg)`

I searched into the web, and I cannot find similar errors for this version of rltk. 

These are the dmesg errors:
```
$ dmesg|grep 88x2bu
[    5.693173] 88x2bu: Unknown symbol ieee80211_get_channel (err -2)
[    5.693222] 88x2bu: Unknown symbol cfg80211_rx_mgmt (err -2)
[    5.693235] 88x2bu: Unknown symbol ieee80211_frequency_to_channel (err -2)
[  132.014996] 88x2bu: Unknown symbol ieee80211_get_channel (err -2)
[  132.015077] 88x2bu: Unknown symbol cfg80211_rx_mgmt (err -2)
[  132.015102] 88x2bu: Unknown symbol ieee80211_frequency_to_channel (err -2)
[  134.699511] 88x2bu: Unknown symbol ieee80211_get_channel (err -2)
[  134.699592] 88x2bu: Unknown symbol cfg80211_rx_mgmt (err -2)
[  134.699617] 88x2bu: Unknown symbol ieee80211_frequency_to_channel (err -2)
[  200.477846] 88x2bu: Unknown symbol ieee80211_get_channel (err -2)
[  200.477984] 88x2bu: Unknown symbol cfg80211_rx_mgmt (err -2)
[  200.478013] 88x2bu: Unknown symbol ieee80211_frequency_to_channel (err -2)
[  229.839239] 88x2bu: Unknown symbol ieee80211_get_channel (err -2)
[  229.839321] 88x2bu: Unknown symbol cfg80211_rx_mgmt (err -2)
[  229.839346] 88x2bu: Unknown symbol ieee80211_frequency_to_channel (err -2)
[  355.035946] 88x2bu: Unknown symbol ieee80211_get_channel (err -2)
[  355.036031] 88x2bu: Unknown symbol cfg80211_rx_mgmt (err -2)
[  355.036056] 88x2bu: Unknown symbol ieee80211_frequency_to_channel (err -2)
[  482.564691] 88x2bu: Unknown symbol ieee80211_get_channel (err -2)
[  482.564772] 88x2bu: Unknown symbol cfg80211_rx_mgmt (err -2)
[  482.564797] 88x2bu: Unknown symbol ieee80211_frequency_to_channel (err -2)
[  490.089092] 88x2bu: Unknown symbol ieee80211_get_channel (err -2)
[  490.089173] 88x2bu: Unknown symbol cfg80211_rx_mgmt (err -2)
[  490.089199] 88x2bu: Unknown symbol ieee80211_frequency_to_channel (err -2)
```
Somebody can tell me what's happening here?

Thanks in advance.

MaxG87, 2021-01-08:
I remember that I got similar problems some time back and I could not find any solution expect sticking around with earlier kernels. Some merge resolved that eventually.

Can you give us the revision you are using? For which kernel do you compile? Does it work with older kernels?

bernhara, 2021-01-08:
Is the "cfg80211" already loaded?

`modprobe cfg80211`

JacksonMaxfield, 2021-01-09:
Having the same issue. My log of `dmesg` is basically the same as above.

Kernel version: `5.8.0-34-generic`
If relevant, Ubuntu version: `20.04.1`

> For which kernel do you compile? Does it work with older kernels?

Yes it works with my `5.4.0-60-generic` kernel.

> Is the "cfg80211" already loaded?

No. Result of `modprobe`:
```
~/src/rtl88x2bu$ modprobe cfg80211
modprobe: FATAL: Module cfg80211 not found in directory /lib/modules/5.8.0-34-generic
```

hdnh2006, 2021-01-11:
> I remember that I got similar problems some time back and I could not find any solution expect sticking around with earlier kernels. Some merge resolved that eventually.
> 
> Can you give us the revision you are using? For which kernel do you compile? Does it work with older kernels?

Hi, 

I am using the version of this repository, my kernel version is `5.3.0-64-generic` for Ubuntu 19.10, I have not tried other kernels.

Any help?

hdnh2006, 2021-01-11:
> Is the "cfg80211" already loaded?
> 
> `modprobe cfg80211`

It seems it is, but I am not sure, when I run the command `modprobe cfg80211`, I do not receive any information but I also do not receive errors, so I guess it is working.

hdnh2006, 2021-01-17:
I found a solution deleting corrupt versions doing the following:

```
sudo dkms remove rtl88x2bu/5.8.7.4 --all
find /lib/modules -name cfg80211.ko -ls
sudo rm -f /lib/modules/*/updates/net/wireless/cfg80211.ko
```

Check if there are other drivers also, if so, you will have to delete them,

closed 2021-01-17 by hdnh2006