cilynx/rtl88x2bu #115: ERROR : Can't compile on RPI4B / Raspbian 64

xMadNet, 2021-01-07:
uname -a : Linux XPiRasp4 5.10.4-v8+ #1389 SMP PREEMPT Wed Jan 6 13:52:18 GMT 2021 aarch64 GNU/Linux
Raspi b4 , Raspbian 64b

=>make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.10.4-v8+/build M=/home/pi/wifidrivers/rtl88x2bu  modules
.....
ERROR: modpost: "__const_udelay" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__arch_copy_from_user" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "_raw_spin_unlock" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__rcu_read_lock" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__rcu_read_unlock" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "cpu_hwcap_keys" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "csum_ipv6_magic" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__warn_printk" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "arm64_const_caps_ready" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__udelay" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "preempt_schedule_notrace" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__arch_copy_to_user" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!

Previous patch poposed by PiGuy314([https://github.com/PieGuy314/RTL88x2BU-RPi4-arm64-Driver-Patch]) doesn't work anymore (manual patch since Makefile modified)
Same erros

Please help
I can test

MaxG87, 2021-01-08:
Could this be a duplicate of #98 ?

xMadNet, 2021-01-08:
Duplicate of  #98 ?
Thanks

closed 2021-01-08 by xMadNet