cilynx/rtl88x2bu #117: Driver doesn't work on Ubuntu 20.04 with the kernel 5.8.0-generic

eroscestlavie, 2021-01-13:
I know a few people have had issues with this already so sorry if I'm repeating anything! The driver stops working when I upgrade to 5.8, but I can't see what the issue is. Applying patch #58 doesn't seem to help. Whenever I reboot in 5.8 my computer is unable to find the network adapter and just comes up with "cable unplugged" for wired connection.

CGarces, 2021-01-13:
patch #58 is already applied so there is nothing to patch.
Try to build again the module from scratch just to ensure that there is no error on your build process.

MaxG87, 2021-01-13:
>I know a few people have had issues with this already so sorry if I'm
>repeating anything! The driver stops working when I upgrade to 5.8, but I
>can't see what the issue is.
Does the driver work when you use `sudo insmod 88x2bu.ko`? Maybe this gives
you a starting point.

eroscestlavie, 2021-01-13:
Oh okay, it's a user error and not an issue then, sorry! I don't know how to build a module, I just followed the instructions for the dkms install.

closed 2021-01-13 by eroscestlavie