cilynx/rtl88x2bu #118: Failed to compile on kernel 5.10.7

StaticallyTypedRice, 2021-01-19:
I'm on Fedora 32, with the following `uname -a`:

```
Linux localhost.localdomain 5.10.7-100.fc32.x86_64 #1 SMP Tue Jan 12 20:25:28 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

When I try to build the module with `sudo ./build.sh`, it returns the following error:

```
make -j4 KERNELRELEASE=5.10.7-100.fc32.x86_64 KVER=5.10.7-100.fc32.x86_64 src=/usr/src/rtl88x2bu-5.6.1...........(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.7-100.fc32.x86_64 (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.6.1/build/make.log for more information.
```

The contents of the `/var/lib/dkms/rtl88x2bu/5.6.1/build/make.log` is:

```
DKMS make.log for rtl88x2bu-5.6.1 for kernel 5.10.7-100.fc32.x86_64 (x86_64)
Mon 18 Jan 2021 11:53:35 PM PST
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.7-100.fc32.x86_64/build M=/var/lib/dkms/rtl88x2bu/5.6.1/build  modules
make[1]: Entering directory '/usr/src/kernels/5.10.7-100.fc32.x86_64'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/linux/usb_intf.o
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c: In function ‘isFileReadable’:
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2203:11: error: implicit declaration of function ‘get_fs’; did you mean ‘get_sa’? [-Werror=implicit-function-declaration]
 2203 |   oldfs = get_fs();
      |           ^~~~~~
      |           get_sa
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2203:11: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2205:3: error: implicit declaration of function ‘set_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
 2205 |   set_fs(KERNEL_DS);
      |   ^~~~~~
      |   sget_fc
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2205:10: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2205 |   set_fs(KERNEL_DS);
      |          ^~~~~~~~~
      |          KERNFS_NS
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2205:10: note: each undeclared identifier is reported only once for each function it appears in
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c: In function ‘retriveFromFile’:
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2245:12: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
 2245 |    oldfs = get_fs();
      |            ^~~~~~
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2247:11: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2247 |    set_fs(KERNEL_DS);
      |           ^~~~~~~~~
      |           KERNFS_NS
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c: In function ‘storeToFile’:
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2284:12: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
 2284 |    oldfs = get_fs();
      |            ^~~~~~
/usr/src/rtl88x2bu-5.6.1/os_dep/osdep_service.c:2286:11: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2286 |    set_fs(KERNEL_DS);
      |           ^~~~~~~~~
      |           KERNFS_NS
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:279: /var/lib/dkms/rtl88x2bu/5.6.1/build/os_dep/osdep_service.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1806: /var/lib/dkms/rtl88x2bu/5.6.1/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.10.7-100.fc32.x86_64'
make: *** [Makefile:2284: modules] Error 2
```

MaxG87, 2021-01-19:
This problem was supposedely fixed in 5.6.1.6 Could you please try out branch 5.6.1.6?
Probably we should make it the default branch soon and abandon the old one.

StaticallyTypedRice, 2021-01-19:
`5.6.1.6_35492.20191025_COEX20180928-6a6a` worked. Thank you!!

closed 2021-01-19 by StaticallyTypedRice