cilynx/rtl88x2bu #122: Switch to 5.6.1.6_35492.20191025_COEX20180928-6a6a as default branch

MaxG87, 2021-01-20:
With the most recent commits 5.6.1.6_35492.20191025_COEX20180928-6a6a should contain every addition and fix that was added to the 5.6.1_... branch. I think its time to make it the main branch.

MaxG87, 2021-01-20:
I just pushed a new branch:  5.8.7.1_35809.20191129_COEX20191120-7777. I think I applied all necessary adaptions on it already.

Lets directly switch to  5.8.7.1_35809.20191129_COEX20191120-7777 please!

MaxG87, 2021-01-29:
Confirmed to work on Debian Bullseye with Kernel 5.10.0-2. Works good enough to post this comment.

```bash
$ uname -a
Linux hostname 5.10.0-2-amd64 #1 SMP Debian 5.10.9-1 (2021-01-20) x86_64 GNU/Linux
```

MaxG87, 2021-01-29:
I overlooked some merge conflict lines. #124 blocks this issue atm.

MaxG87, 2021-03-08:
#124 is resolved. Please lets reconsider to switch to 5.8.7.1 as default branch.

AGenchev, 2021-04-10:
I guess he didn't switch because per issue #144 it didn't compile on RPI....

MaxG87, 2021-04-16:
Thank you for bringing #144 to my attention. In the meantime I created #148 , which should solve that issue.

AGenchev, 2021-04-16:
Thank you for working on it!

cilynx, 2021-09-04:
Default branch has been updated to 5.8.7.1_35809.20191129_COEX20191120-7777

MaxG87, 2021-09-06:
Thank you very much.

closed 2021-09-06 by MaxG87