cilynx/rtl88x2bu #124: Merge issue

ryleyb, 2021-01-25:
Looks like some merge stuff got accidentally committed here...

https://github.com/cilynx/rtl88x2bu/blob/ba12c4f939948c1b16db7276da94feeef294da6c/os_dep/linux/rtw_proc.c#L5072

MaxG87, 2021-01-29:
Thank you for noticing. The error was introduced in 822ca8a494024231b6798a0eaff208b7f02a4d43 where I cherry picked 4 individual commits. The commits are named in the commit message.

Resolving their merge conflicts is quite tough. I would welcome a helping hand here but I will retry cherry-picking the entire thing soon on my own too.

MaxG87, 2021-03-08:
I fixed the merges with the most recent force-push. Please open a new issue in case of any problems.

closed 2021-03-08 by MaxG87