cilynx/rtl88x2bu #126: lib module issue

GarethHards, 2021-02-02:
pi@raspberrypi:~/usbwifi/rtl8821CU $ sudo dkms build -m rtl88x2bu -v ${VER} # Takes ~3-minutes on a 3B+
Error! echo
Your kernel headers for kernel 5.10.11-v7l+ cannot be found at
/lib/modules/5.10.11-v7l+/build or /lib/modules/5.10.11-v7l+/source.
pi@raspberrypi:~/usbwifi/rtl8821CU $ sudo dkms install -m rtl88x2bu -v ${VER}
Error! echo
Your kernel headers for kernel 5.10.11-v7l+ cannot be found at
/lib/modules/5.10.11-v7l+/build or /lib/modules/5.10.11-v7l+/source.

i get this issue and i went to the lib module folder: and i have a few listed the newest is 5.10.11-v8+ 
so what do i do now to get the install to work my onboard wifi is still active and working. but i want to put on raspap

GarethHards, 2021-02-02:
sorry didnt mention ubuntu 19.04 on a pi4 b 4gb armhf 64 running 32 bit rasberian

MaxG87, 2021-05-31:
Have you installed the Linux kernel headers? I do not know how the package would be named on Ubuntu for RPI, tough.