cilynx/rtl88x2bu #129: Ubuntu 20.04 and Kernel 5.8.0-41-generic not working

menwn, 2021-02-26:
when boot the Ubuntu machine with 5.8.0-41-generic kernel the Wifi is not shown and I cannot make it work although I tried to re-run the installation steps.

Anyone had this issue? Is it something that we could do?

markybry, 2021-02-26:
I manage to get it to work with 5.8.0-44-generic Ubuntu 20.10 . However, when I ran "sudo modprobe 88x2bu" I received an error stating the action was not permitted. I had an issue with a secure boot which I resolved by temporarily disabling it by doing the following. https://askubuntu.com/questions/1114867/operation-not-permitted-when-trying-to-modprobe-xpad
Install mokutil "sudo apt-get install mokutil"

and check the status of SecureBoot "mokutil --sb-state"

In case it is enabled run command "sudo mokutil --disable-validation"

Now enter a temporary password between 8 to 16 digits. We will use this password later. Enter the same password again to confirm. Once it’s done reboot the system and press any key when you see the blue screen (MOK management). Select Change Secure Boot state. Enter the password you had selected before and press Enter. Select Yes to disable Secure Boot in shim-signed. Press Enter key to finish the whole procedure.

MaxG87, 2021-03-08:
Thank you for your report.

closed 2021-03-08 by MaxG87